ArduPilot/ardupilot #12227: UBLOX UBX-M8030 GPS no fix

Bumerangas, 2019-09-06:
### Bug report

**Issue details**
No GPS fix. With u-center_v19.08.01 via FTDI all working well.

**Version**
ALL

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
Kakute F7 mini

**Logs**
u-blox 1 HW: 00080000 SW: ROM CORE 3.01 (107888)
EKF2 IMU0 tilt alignment complete
Frame: QUAD
KakuteF7 00260033 3138510E 35353931
ChibiOS: d2030d88
ArduCopter V3.6.9 (632be63f)
GPS 1: detected as u-blox at 115200 baud
PreArm: RC Yaw not configured
PreArm: RC Throttle not configured
PreArm: RC Pitch not configured
PreArm: RC Roll not configured

arjun7965, 2019-09-09:
Did you check the number of satellites, HDOP and other GPS related  data?

WickedShell, 2019-09-09:
This is a support issue, please open a discussion on discuss.ardupilot.org. It's most likely a cabling problem that is preventing the autopilot from configuring the GPS.

closed 2019-09-09 by WickedShell

Bumerangas, 2019-09-09:
No it is not cabling, as you can see from logs module detected, HDOP 100 etc. FC sending config if it is disabled anyway or maybe module resenting it every start. Try this GPS module https://www.team-blacksheep.com/products/prod:tbs_m8_gps_glonass