ArduPilot/ardupilot #12251: Rover: Add true CRUISE mode, primarily for Boats

Hwurzburg, 2019-09-10:
### Feature request
Setting up my first Rover , actually a boat, I was immediately struck that there was no CRUISE mode like in Plane....ie running a straight GPS course without any pilot input, and resetting it once  the pilot stops inputting heading changes....

this seems a natural mode for a boat....just like full size boat autopilots...hold a straight GPS course..

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x ] Rover
[  ] Submarine

rmackay9, 2019-09-10:
I think handling the desired heading could make this mode difficult to understand for users.  Boats can stop (which planes of course can't) so the heading required to keep the vehicle moving in the desired direction could be very different from the "desired heading" (input by the pilot's yaw stick).

I was thinking that simple mode might be a more intuitive way for users to achieve something similar

Hwurzburg, 2019-09-10:
I dont understand the problem.....if it can navigate in AUTO, it should be able to in CRUISE....and whether or not its moving when course is being set or reset is not relevant for plane either, as I understand it.....when the pilot stops changing the actual heading of the vehicle with input (ail OR rudder), the new waypoint is set 1KM out....and keeps getting updated 1KM out every time period (I think .5 sec, but should probably be longer for rover since they are slower...maybe every 2-3s) as long as its in mode and no pilot input occurs...if the pilot is stationary, when mode is entered or input stops, it just starts cruising in the current heading direction......I dont understand why this would not work on a boat...

rmackay9, 2019-09-11:
.. but what about controlling the vehicle's speed?  I think it would be weird for a user to have a mode where they can't control the speed... and then if you suggest that the speed be controlled with the throttle stick then let's consider the vehicle's heading when it's in a river and the user moves the throttle stick to the middle so it stops.  and then how the vehicle responds when they move the throttle forward or backwards (to reverse).

Hwurzburg, 2019-09-11:
the plane analogy I think still applies....current = wind.....optionally in plane, in cruise mode, below  1/2 throttle (above 3/4 throttle in rover I would suggest), a user set cruise throttle would be applied.....above, it increases throttle (if no airspeed sensor, with sensor, it increases target airspeed)....if wind is greater than vehicle speed at any point, it maintains heading trying to get to next waypoint....but drifts backward...unless you can increase speed with increased throttle....

without that throttle option enabled, it just uses a fixed cruise throttle set by user...if the wind is too much, too bad

again, how does the boat get to the next waypoint in AUTO in a river?
I wouldnt expect CRUISE to operate the same way....

rmackay9, 2019-09-11:
ok, at this point we probably need to discuss verbally.  The key point I'm trying to make is how to handle the case of the user wanting the vehicle to stop by bringing the throttle stick back to center.

It's not a question of technical programming difficulty really.

rmackay9, 2019-09-11:
After a discussion with @tridge and @Hwurzburg we decided on the mode being like this:

- pilot's throttle controls the speed of the vehicle
- pilot's steering input primarily controls the vehicle's desired ground heading but to provide immediate feedback to the user it also controls the vehicle's rotation rate (like acro).  When the steering input is at zero the vehicle attempts to maintain the desired ground heading (using the L1 controller)
- when both throttle and steering input are zero the vehicle attempts to loiter.

we also discussed the need for a "reverse throttle" RC switch but that will be created as a spearate issue