ArduPilot/ardupilot #12259: Nothing happened when onboard mag calibration is started.

LMD93, 2019-09-10:
**Issue details**
Hi, I am trying to use hereflow for my quadcopter. Im using a purple cube. Initially I went with firmware ArduCopter version 3.6.10 (stable). However in the full parameter list on Mission Planner, the param 'Flow_Type' is missing so I could not enable the hereflow.

So I tried firmware ArduCopter 3.7 (latest), which I was able to get the hereflow working. However, when I tried to run the onboard mag calibration, nothing happens as shown below. I tried to calibrate my compass on QGroundControl as well but same thing happened, which is nothing. 

There isn't any errors thrown. 

**Version**
ArduCopter 3.7

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Cube Purple

**Logs**

[image]

rmackay9, 2019-09-11:
Can you try resetting the FLOW_TYPE to zero and re-doing the compass calibration?  I'd like to separate whether the Here flow is related to the compass issue.  i.e. does it work without the HereFlow?

LMD93, 2019-09-11:
Hello @rmackay9, I have tried changing the FLOW_TYPE to zero but compass calibration is still not working. I have also tried repeating the same steps on Cube Black, it is not working as well.

rmackay9, 2019-09-11:
compass calibration is working on Copter-3.6.10 and 3.7-dev so I'm afraid at this point that it's a support issue and so I'll close this issue and it would be best to raise the issue in [the support forums](https://discuss.ardupilot.org/c/arducopter/copter-36).

Sorry to be strict about this but otherwise our to-do list just turns into another support forum.

P.S. there is some advice in the "If calibration fails" section of the [compass calibration wiki page](http://ardupilot.org/copter/docs/common-compass-calibration-in-mission-planner.html).

closed 2019-09-11 by rmackay9