ArduPilot/ardupilot #12260: Arduplane versions newer than 3.9.8 do not load on Parrot CHUCK

mainframe, 2019-09-10:
### Bug report

**Issue details**

Arduplane builds newer than 3.9.8 do not load on Parrot CHUCK. It seems to require newer glibc (v2.27) than available on CHUCK (Parrot firmware 1.7.1, glibc v2.19). 

**Version**
3.9.10

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
Parrot CHUCK, Disco fw 1.7.1

**Logs**

lucasdemarchi, 2019-09-20:
It is a toolchain problem, not really ardupilot. I think I have instructions to share somewhere.... I will try to find and link them here

lucasdemarchi, 2019-09-20:
So, what I do is to use buildroot to side load binaries in disco. This allows me to avoid static linking, use libiio, ardupilot modules, etc (modules are not available if you static link because you can't use dlopen().

Basically what you need is to download buildroot, configure it and then add this file and link:
`host/bin/arm-linux-gnueabihf-pkg-config -> pkg-config`

```
$ cat ~/p/buildroot/output/host/bin/arm-linux-gnueabihf-pkg-config 
#!/bin/sh
PKGCONFDIR=$(dirname $0)
DEFAULT_PKG_CONFIG_LIBDIR=${PKGCONFDIR}/../arm-buildroot-linux-gnueabihf/sysroot/usr/lib/pkgconfig:${PKGCONFDIR}/../arm-buildroot-linux-gnueabihf/sysroot/usr/share/pkgconfig
DEFAULT_PKG_CONFIG_SYSROOT_DIR=${PKGCONFDIR}/../arm-buildroot-linux-gnueabihf/sysroot
PKG_CONFIG_LIBDIR=${PKG_CONFIG_LIBDIR:-${DEFAULT_PKG_CONFIG_LIBDIR}} PKG_CONFIG_SYSROOT_DIR=${PKG_CONFIG_SYSROOT_DIR:-${DEFAULT_PKG_CONFIG_SYSROOT_DIR}} exec ${PKGCONFDIR}/pkgconf  "$@"
```

I'm not sure, but I think the SYSROOT_DIR variable is not set by default in buildroot. This way I can build for disco as for any other board:
```sh
./waf configure --board disco
./waf plane
```

Configure your buildroot to add libiio and you should see something like this:

```
Checking for program 'arm-linux-gnueabihf-pkg-config' : /home/ldmartin/p/buildroot/output/host/bin/arm-linux-gnueabihf-pkg-config 
Checking for 'libiio'                                 : yes 
Checking link with libiio                             : yes 
```

To confirm it won't require libc symbols with versions greater than the available on disco:
```
$ arm-linux-gnueabihf-objdump -T build/disco/bin/arducopter | grep GLIBC_ | awk '{ print $5 }' | sort -u
00000004
GLIBC_2.12
GLIBC_2.17
GLIBC_2.4
GLIBC_2.7
GLIBC_2.8
GLIBC_2.9
```

This is also what I do to add other binaries in disco or the skycontroller. See my blog post about using evemu for example: https://politreco.com/2019/04/using-evemu-to-simulate-input-devices/ - I also use it to build dema-rc and load in the SkyController2: https://github.com/lucasdemarchi/dema-rc (last months I went short of time and couldn't complete this project though... idea is that it will allow you to control ardupilot in disco using its normal controller).

I had a draft of a blog post about using buildroot to side load binaries in disco, but I lost it while changing my computer. I will try to write it again soon, but I hope the instructions above help.

lucasdemarchi, 2019-10-04:
Oh, the blog post is here: https://politreco.com/2019/09/side-load-on-embedded-linux-with-buildroot/

let me know if that works for you so we can close this PR.

mainframe, 2019-10-05:
Will try it and let you know. Thanks for the info!

mainframe, 2020-09-15:
Here are the instructions for latest Arduplane build for Disco:
```bash
# install buildroot
apt install unzip
cd /opt
curl -L -o toolchain-master.zip https://gitlab.com/lucas.de.marchi/dema-rc/-/jobs/545230428/artifacts/download
unzip toolchain-master.zip
unxz toolchain.tar.xz
tar xvf toolchain.tar

# prepare arduplane build env
cd /opt
git clone https://github.com/ArduPilot/ardupilot.git
cd ardupilot/
git fetch && git fetch --tags
git checkout ArduPlane-4.0.6
git submodule update --init --recursive
Tools/environment_install/install-prereqs-ubuntu.sh -y
. ~/.profile

# supply pkg-config from buildroot
cd /usr/bin
# you may skip this line if default arm-linux-gnueabihf-pkg-config is not installed in the system
mv arm-linux-gnueabihf-pkg-config arm-linux-gnueabihf-pkg-config.orig
ln -sf /opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/pkg-config arm-linux-gnueabihf-pkg-config
vim /opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/pkg-config
--- MODIFY ---
#PKGCONFDIR=$(dirname $0)
PKGCONFDIR=/opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin
--- MODIFY ---

# build arduplane
cd /opt/ardupilot
export PATH=/opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/:$PATH
./waf configure --board disco
./waf plane
```

closed 2020-09-15 by mainframe

lucasdemarchi, 2020-09-15:
> # supply pkg-config from buildroot

Don't do this as it will stop working when the package in your distro is updated. Instead, just export PATH to have /opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/ earlier so it uses pkg-build from there rather than from /use/lib

mainframe, 2020-09-15:
It did not work in my case - waf configure still picked /usr/bin/arm-linux-gnueabihf-pkg-config - regardless of setting export PATH=/opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/:$PATH

mainframe, 2020-09-15:
Ok found the solution - I was missing /opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/arm-linux-gnueabihf-pkg-config -> pkg-config linkage - now path precedence works as intended.

lucasdemarchi, 2020-09-15:
Humn... I thought that was already in the toolchain tarball... Probably something to be added after the relocate call: https://github.com/ArduPilot/dema-rc/blob/master/ci/Dockerfile.debian#L33

hsy77101, 2025-04-10:
Dear lucasdemarchi and mainframe

So, did you solve this issue?
Actually, I am trying to load newer version than 3.9.8 to Parrot Disco Chuck.
I have no idea how to load it exactly.
Can you let me know full instructions for latest Arduplane build for Disco?

hsy77101, 2025-04-10:
> Ok found the solution - I was missing /opt/arm-buildroot-linux-gnueabihf_sdk-buildroot/bin/arm-linux-gnueabihf-pkg-config -> pkg-config linkage - now path precedence works as intended.

Dear mainframe

So, did you solve this issue?
Actually, I am trying to load newer version than 3.9.8 to Parrot Disco Chuck.
I have no idea how to load it exactly.
Can you let me know full instructions for latest Arduplane build for Disco?