ArduPilot/ardupilot #12264: Improved Failsafe Handling During Auto-Landings

SUI-Eric, 2019-09-11:
#### Issue details
If an aircraft is performing an automatic landing (after a DO_LAND_START) and encounters a failsafe with an action of LAND, it will restart the landing sequence from the beginning. If - for example - an aircraft encounters a critical battery failsafe on final approach, the resulting aborted landing and go-around can be especially taxing on the already-low battery. Suggest that failsafe behavior be updated such that if the aircraft is already performing a landing sequence and encounters a failsafe with LAND action (or perhaps also RTL with RTL_AUTOLAND 1) that it simply continue the approach rather than revert to DO_LAND_START. 

#### Version
AP3.9.8

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Fixed-Wing Airplane

#### Hardware type
Cube Black

#### Logs
Can share some log excerpts upon request.

Gone4Dirt, 2019-12-17:
Hi have put in a PR.  Hope to resolve this issue soon.

IamPete1, 2020-02-18:
closed by https://github.com/ArduPilot/ardupilot/pull/13065 for basic missions this should now be fixed, if you have complex missions where there are jumps after the do land start you will need to enable distance based caculation with MIS_OPTIONS bit 1, ie + 2 to what you have set

closed 2020-02-18 by IamPete1