ArduPilot/ardupilot #12293: SITL Plane treats SERVOn_FUNCTION = 0 as SERVOn_OUT = 0

vierfuffzig, 2019-09-14:
Stumbled across the plane flying funny squiggles if rudder servo is disabled (SERVO4_FUNCTION = 0). It doesn't if i leave it at 21 and fix pwm output to 1500. My guess is SITL treats PWM 0 as a valid output here.

SITL ArduPlane V3.10.0-dev (b4537beb)

Log: https://github.com/vierfuffzig/Chibios-miscellaneous/raw/master/SITL_OUTPUT_0.BIN

peterbarker, 2019-09-16:
Please check again in master.

I broke it and tridge reverted that commit.

vierfuffzig, 2019-09-16:
@peterbarker thanks for looking into this. 
tested on latest commit https://github.com/ArduPilot/ardupilot/commit/8b92df51 as well as  https://github.com/ArduPilot/ardupilot/commit/6c6a7cc prior to your respective commit https://github.com/ArduPilot/ardupilot/commit/1735563 , the issue persists in both, so imho this is unrelated.
on a sidenote, it requires restarting SITL after you set SERVO4_FUNCTION = 0 to show the effect. setting it back to active however solves the squiggly flight pattern promptly on param write.

marck3131, 2022-01-13:
Hello @peterbarker , is this issue resolved yet? I could see no responses in this thread , Can I work on this issue? 
Thankyou.

hendjoshsr71, 2022-01-13:
If you are interested in it go ahead! First PR to be submitted and merged wins!

peterbarker, 2023-01-31:
Reproduction is straight-forward.

First, see normal behaviour with:
```
mode fbwa
arm throttle
rc 3 2000
```

takes off in a straight line

then disable the rudder servo:
```
param set SERVO4_FUNCTION 0
```

then try the same takeoff again.  You should see the thing essentially turning on the spot.  That's not what it should do.

jrcacador, 2025-03-07:
Hi! I'm interested in solving this issue. Can I work on it?

PR #29652 (2025-04-01): SITL: handle case where servo is disabled in servo angle filtering
Fixes #12293. If the servo is disabled (value 0), assign the default trim value of 1500 before applying filtering.

In the filtered_servo_angle function of the Aircraft class, servo signals are expected to range from 1000 to 2000, with a neutral trim position at 1500. However, when a servo is disabled (value 0), the function incorrectly processes it as a valid PWM value, leading to unexpected behavior.

This fix ensures that disabled servos are correctly handled by explicitly setting their value to 1500 before filtering. As a r [... cut]
closed 2025-04-01 by tridge