ArduPilot/ardupilot #12295: Rover: FS failure

Hwurzburg, 2019-09-14:
### Bug report

**Issue details**

When completing a mission with no RTL as last command...the mission goes into HOLD mode ...unfortunately, if FS occurs while in RTL or HOLD, it is ignored...the vehicle will not RTL or SMARTRTL....the FS entry code should ignore RTL or SMARTRTL modes, but not HOLD mode

line 77 in failsafe.cpp code should be changed, I think...

**Version**
3.5 Stable
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

khancyr, 2019-09-14:
That isn't a bug. We decided this behavior as it is better to have the rover/ wait at known place than at a random place during RTL.

khancyr, 2019-09-14:
Many we could make this a parameter

Hwurzburg, 2019-09-14:
I dont understand this at all....the purpose of FS actions like RTL and SmartRTL is to retrieve the vehicle in case of RC/GCS/Batt loss....I am not arguing that end of mission default should not be HOLD (although LOITER would be better if its a boat frame).....but FAILSAFE should bring the vehicle home....why else have FS_ACTION??? if you want HOLD you can select it in that param....but if you say #3 SmartRTL/RTL there should be no case where FS does not deliver that action...

khancyr, 2019-09-15:
No failsafe should prevent harmful situation for the drone or human, not enforcing going back to home.

And you are wrong, all other vehicle are doing the same, if you are landed or close to land, it won't put gas and get back to rtl. They will land or disarm.

Case for RC and GCS failsafe could be different as they aren't critical for the vehicle, even if we can debate that is could be dangerous to let a vehicle move on ground at high speed without way to control it.

Hwurzburg, 2019-09-15:
the same situation will occur with any FS  in any other mode..it could  RTL...should all RTLs be prevented?

perhaps, this should be limited to boat frames? I put a PR in that we can discuss at the DEVCALL

rmackay9, 2019-09-16:
By the way, there is a parameter that changes a boat into Loiter mode at the end of a mission instead of Hold.

Hwurzburg, 2019-09-16:
After thinking about this, I think Pierre's suggestion of leaving everything as is, but adding a HOLD_FS_BEHAVE option to allow or prevent FS initiation while in HOLD mode....I will try to code as an alternative before DEVCALL...

Hwurzburg, 2019-09-17:
After Dev call decided to use the above approach of a param to enable/disable entry in FS while in HOLD

Hwurzburg, 2019-09-22:
PR entered

closed 2019-09-22 by Hwurzburg