ArduPilot/ardupilot #12322: AP_Arming:AP_Proximity: proximity pre-arm check failure if using only upward facing lidar

rmackay9, 2019-09-17:
If a copter is setup with only an upward facing lidar (i.e. no horizontally facing lidar) and PRX_TYPE = 4 (i.e. proximity sensor enabled using range finders) then a "PreArm: check proximity sensor" message will appear and the vehicle cannot be armed (unless arming checks are disabled).  [The pre-arm check code can be seen here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_Arming.cpp#L710).

The underlying issue is the AP_Proximity_RangeFinder class (the proximity library's backend that pulls data from rangefinders) updates it's status based on horizontally facing lidar only ([see code here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Proximity/AP_Proximity_RangeFinder.cpp#L77)) and ignores upward facing lidar.

The solution may be to add an AP_Proximity::get_upward_status() that checks only upward facing lidar and modify the existing AP_Proximity:get_status() to only apply to horizontal proximity

peterbarker, 2020-01-07:
@rmackay9 Pretty sure you did this one.

rmackay9, 2020-01-07:
@peterbarker, i'm not sure actually, maybe!

IamPete1, 2021-08-29:
closed by https://github.com/ArduPilot/ardupilot/pull/17781

closed 2021-08-29 by IamPete1