ArduPilot/ardupilot #12327: AP_BattMonitor I2C abstraction on Pixhawk 4

nbelanger99, 2019-09-17:
### Bug report

**Issue details**

Debugging the AP_BattMonitor library, and at least for the Pixhawk 4, neither directive:

```
#define AP_BATTMONITOR_SMBUS_BUS_INTERNAL           0
#define AP_BATTMONITOR_SMBUS_BUS_EXTERNAL           1
```

Refers to the actual "I2C A" port on the board. Apparently 3:

`hal.i2c_mgr->get_device(3, AP_BATTMONITOR_SMBUS_I2C_ADDR, 100000, true, 20));`

Is the appropriate bus. 

**Version**
3.7

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk 4

nbelanger99, 2020-05-04:
Update: this is still an issue in current master branch. 
See pull request https://github.com/ArduPilot/ardupilot/pull/14284

tridge, 2020-11-10:
it's not actually an issue, it is a hangover from the days when boards had only 2 I2C buses.
For boards with more than 2 I2C buses you must set the BUS parameter on the instance

WickedShell, 2021-04-13:
Closing as these days we have a bus parameter for I2C batteries to resolve this for you.

closed 2021-04-13 by WickedShell