ArduPilot/ardupilot #12338: SmartPort telemetry doesn't work with serial option 4 on Matek F405 Wing

tenten8401, 2019-09-18:
### Bug report

**Issue details**
I have an FrSky R-XSR's SmartPort (pre-inverted) wire hooked up to TX1 on my Matek F405-Wing. I have the following options set:
[image]
As I understand it, SERIAL1_OPTIONS 4 should enable half-duplex mode and it should "just work" since it's already been pre-inverted by the R-XSR itself, however it does not. A similar setup (softserial & halfduplex) works as expected in iNav, so I know it's not a hardware flaw.
I've also tried setting options to 7 and running with uninverted signal, which doesn't work at all either.

I am running the Yaapu telemetry script on my radio and I've also tried regular S.Port protocol 4 to no avail.

**Version**
3.10.0-FIRMWARE_VERSION_TYPE_DEV
Nightly from 9/17/19

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Sonicmodell AR Wing

**Hardware type**
Matek F405-Wing

**Logs**
N/A

IamPete1, 2019-09-18:
F405's don't support serial option's its F7/H7's only

tenten8401, 2019-09-18:
I was under the impression it works since Matek themselves said they had it working on the F405-Wing with option 7 and no inversion done: https://github.com/ArduPilot/ardupilot/issues/9267#issuecomment-463897884

IamPete1, 2019-09-18:
My mistake, I checked in the code, F405 can do half duplex but only F7/H7 can do the inversion and swapping. How have you got it wired up exactly?

tenten8401, 2019-09-18:
I have VCC to 4.5v (on with USB), inverted sbus to rx2 and inverted smartport to tx1. Sbus works fine without issue.

IamPete1, 2019-09-18:
try a different serial port for smartport. I'm not sure if half duplex means you should use tx or rx or if it matters.

tenten8401, 2019-09-18:
I tried 1, 4 and 6 all on tx earlier, but I'll try them again when I get home. I believe half duplex does receiving and transmitting on the tx pin.

Target0815, 2019-09-18:
> it's already been pre-inverted by the R-XSR itself [...]
Only on the soldering pad "P". Did you use this?

tenten8401, 2019-09-18:
I tried soldering to the P pad on there as well, couldn't seem to make it work at all, but I can try it again when I get home. Under iNav it works with the regular S.Port on the connector though so I'm pretty sure that's the pre-inverted one

[image]

tenten8401, 2019-09-18:
Still no telemetry using the "P" pad with SERIALX_OPTIONS = 7 on TX1, RX1, TX6 or RX6 using both regular S.Port & Passthrough, adjusting parameters as recommended for each:
[image]
(tried serial1 for rx1/tx1 along with S.Port & passthrough (protocol 4 & 10) and serial option 7)

Target0815, 2019-09-19:
This **must** work like this ... Soldering point "P" to TX1 or TX6 and SERIALx_OPTIONS=7.

You have used the current master?

Target0815, 2019-09-19:
I successfully tested it today with an RX6R. The solder point "P" is connected to TX6 on the F405-Wing, SERIAL6_x is set to 57 / 7 / 4 and the values appear immediately on the telemetry screen of an X9D.

tenten8401, 2019-09-19:
I'm still seeing the same behavior with my R-XSR. These pictures were taken directly after flashing http://firmware.ardupilot.org/Plane/2019-09/2019-09-19-12:09/MatekF405-Wing/arduplane_with_bl.hex using Betaflight Configurator and then loading my parameters with QGroundControl. The firmware on my R-XSR is the latest non-F.Port firmware: https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/R-XSR/FW-RXSR-190311.zip

My ArduPilot parameters file is here: [ar-wing-latest.params.zip](https://github.com/ArduPilot/ardupilot/files/3631501/ar-wing-latest.params.zip)

My only guess at this point is there's some variance in the firmware between your RX6R and my R-XSR, or my board somehow has 2 defective serial ports.

Here are some pictures of my setup:
[image]
[image]
[image]
[image]
Nothing else is connected to the flight controller.

IamPete1, 2019-09-19:
you could try the Frsky pass-through and Yaapu telem script, that gets more testing than the direct pass-through. 

https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443

tenten8401, 2019-09-19:
The Yaapu telemetry script is officially supported on my radio, but telemetry on that doesn't work either with protocol set to 10 under OpenTX 2.2.4...
[image]
[image]

Target0815, 2019-09-19:
@IamPete1 Your recommendation misses the point. Standard S.Port telemetry should already work ...

Target0815, 2019-09-19:
@tenten8401 Everything looks right to me. I look afterwards again whether I find a R-XSR ...

IamPete1, 2019-09-19:
I'm not sure what else we can suggest, you defiantly bound to the receiver with telemetry enabled? You should get (on taranis anyway) 'telemetry lost' voice message if you turn off the receiver. I have found its quite easy to get poor connections on the wing's pin headers, I have had more luck soldering to the board directly although that could be a massive coincidence over more than one serial port. Are you powering over USB only? have you tested on battery power?

tenten8401, 2019-09-19:
I've tried battery power and USB power and I do get the "telemetry lost" / "telemetry recovered" voice messages along with RSSI and rx voltage transmitted back in the telemetry screen.

Target0815, 2019-09-19:
@tenten8401 I can confirm that it does NOT work in your configuration with an R-XSR. Either the signal on the "P" solder pad of the R-XSR is different or there is no signal. 

If I connect a converter to the normal S.Port output, the telemetry with the R-XSR works fine. My recommendation is to use a converter to avoid this tiny soldering work.

tenten8401, 2019-09-19:
@Target0815 So you're taking the Regular S.Port (inverted) output and then running it through a converter? Is the converter inverting it again or what's happening there?

I'd really prefer to avoid using any more equipment than I have to, especially since it seems to just work for the most part under iNav without much fiddling required.

Target0815, 2019-09-19:
The normal S.Port output is not inverted (signal at scope: 0-3.3V). On the "P" solder pad the signal is inverted (0 to -3.3V). I know that is confusing ...

Take this converter, it will work with any FrSky receiver at the normal S.Port output: https://www.banggood.com/custlink/vDDGosmMEH

You can't compare this with iNav, by the way. Here the serial ports are managed completely different internally. In practice I find this much better with iNav (and Betaflight) and also "more modern" when it comes to newer hardware. But this is about ArduPilot ...

tenten8401, 2019-09-19:
So wait, if the R-XSR has a built-in inverter and I have access to both the uninverted and inverted pads, then why do I need an external inverter? Would it not just be getting inverted twice then? Am I missing something here? I see both TX and RX coming out which makes me think all I really need is a diode for sending both signals down 1 wire since the RS232 converter in this circuit (built into the R-XSR as I understand it) is just inverting it:
(found in a youtube video)
[image]

I might try a diode first and if that doesn't work I'll just give in an order the inverter.. would be really nice not to need it though

Target0815, 2019-09-19:
Can you read German? But you can also recognize it from the circuit diagrams. The easiest way is with a 74LVC2GU04 plus diode. This is also on the converter of Banggood.

[S.Port Inverter.pdf](https://github.com/ArduPilot/ardupilot/files/3632448/S.Port.Inverter.pdf)

tenten8401, 2019-09-19:
Hmm, I might try building one if I can get the parts faster than Banggood.

I still don't see why an external inverter is necessary though, I feel like this might be a case of a slightly different signal that's not getting handled correctly or something.. The "P" port should be a normal uninverted signal because I remember having to use that for F.Port support since F.Port is an uninverted protocol.

tenten8401, 2019-09-26:
Just got a little ebay inverter in similar to the one in the diagram I linked above, soldered it together with the diode, still zero telemetry on protocol 4 or 10. I also tried both with options 0 & 4 for serial6 because of the diode and still no go :(

IamPete1, 2019-09-26:
the diode does the job of converting to half duplex, so you should not need the serial option with it. I will see if I can get it working without, I have a copter who's converter seems to have gone on the blink.

tenten8401, 2019-09-26:
Yeah, that's what I figured it was doing. I couldn't make it work with half duplex or without on both the uninverted and inverted port connected to the inverter, tried swapping rx/tx on the inverter too to no avail.

tenten8401, 2019-10-01:
So I got it working properly with a little converter thing on Amazon, however I still feel like I should leave this issue open because you still need additional hardware for whatever reason while other software manages to do it fine without.. I don't know what it's actually doing internally that would cause it to even need additional hardware in the first place but I feel like something's not working the way it should (that half duplex option maybe, since the uninverted port should work as expected on the R-XSR according to many people)

manotroll, 2019-10-15:
I have mateksys f405-std
And now with the R-XSR so if I buy this converter from banggood will telimetry work without problem?
I used the converter on px4 that I built myself and it works would work here too?
who is working could show pictures?

manotroll, 2019-11-01:
did anyone get any solution for the f405 std?
it seems that 4.0 is no longer beta but still does not work
if adding an Arduino could solve?

tenten8401, 2019-11-01:
@manotroll You will need the hardware converter, since the hardware on the F405 is not capable of inverting it on the fly and ArduPilot does not have SoftSerial support. Here is the convert on Banggood: https://www.banggood.com/FrSky-Yaapu-Telemetry-Converter-Cable-Pixhawk-to-Horus-X10S-X12S-Taranis-X9DP-QX7X7S-Smart-Port-Receiver-X8R-X4R-XSR-RS8R-p-1535564.html?rmmds=search

I think I'm going to close this issue as a request for softserial would probably be a better issue.

closed 2019-11-01 by tenten8401

manotroll, 2019-11-03:
thanks i bought this adapter to test but the signal on the output of the max3232 is reversed i think
although i bought a different model
I couldn't use an Arduino for this?
i have some mini here
this signal at 1 photo and before max3232 and at 2 and after max3232

https://i.imgur.com/wgwDvGP.png
https://i.imgur.com/5BjT9yK.png

dposea, 2019-11-05:
I'm having almost the same problem. I ordered 10 MAX232 chips, and I have built inverters using 1N4001 and 1N4007 diodes, some with a diode on the power in to drop the voltage, some with an inline resistor, and one with a 2K/1K resistor bridge to get 3.3V instead of 5. I followed the wiring on the "Soldering" blog exactly, but I never get any telemery discovered. I have a scope coming tomorrow. I can't find any other way to find out what is not working. I have tried every UART/Serial port. Assuming I want to try using the uninverted s.port signal, I wire the P test point on the R-XSR to the TX? pin, and set the parameters to 57,7,10? Do I need a diode between TX and RX on the F405?

Target0815, 2019-11-05:
The MAX232 chips are rather unsuitable, because much too high levels are given to the S.Port. It works best with this converter: https://www.banggood.com/custlink/vDDGosmMEH

There a 74LVC2GU04DBVR 2x-inverter is installed, which delivers standard-compliant levels.

Further possibilities for functional converters can be found in the attached .pdf (the text there is in German, but the circuit diagrams should be clear).

[RS232_Inverter.pdf](https://github.com/ArduPilot/ardupilot/files/3807854/RS232_Inverter.pdf)

With these converters the F405-Wing works without any special features, here a UART is directly connected to the S.Port of a receiver:

**F405-Wing with converter:**
SERIALx_BAUD=57
SERIALx_PROTOCOL=4 (S.Port) oder 10 (S.Port Passthrough /Yaapu)
SERIALx_OPTION=0

If **no converter** is to be used, a "hacked" RX **must** be used. For example the mentioned R-XSR with the P-Pad:

**F405-Wing without converter, hacked S.Port:**
SERIALx_BAUD=57
SERIALx_PROTOCOL=4 (S.Port) oder 10 (S.Port Passthrough /Yaapu)
SERIALx_OPTION=4 (oder 7)
(SERIAL4 or SERIAL6)

**F765-Wing without converter, normal S.Port (in this example connected to TX1):**
SERIAL2_BAUD=57
SERIAL2_PROTOCOL=4 (S.Port) oder 10 (S.Port Passthrough /Yaapu)
SERIAL2_OPTION=7

manotroll, 2019-11-05:
> I'm having almost the same problem. I ordered 10 MAX232 chips, and I have built inverters using 1N4001 and 1N4007 diodes, some with a diode on the power in to drop the voltage, some with an inline [...]

I bought the indicated inverter from the post below
I must use inav until my inverter arrives this one of the bc548 I had already done and it didn't work
could not in future be added softserial as well as no inav?

dposea, 2019-11-05:
I had no luck with this either:
F405-Wing without converter, hacked S.Port:
SERIALx_BAUD=57
SERIALx_PROTOCOL=4 (S.Port) or 10 (S.Port Passthrough /Yaapu)
SERIALx_OPTION=4 (or 7)
(SERIAL4 or SERIAL6)

Does this only work on SERIAL4 or SERIAL6? I was using SERIAL1. Should the OPTION be 4 or 7?
It's very difficult to find accurate, up to date info on this issue. There was an issue on Ardupilot Github but no one really ever explained what is now possible for F405 Wing.

Target0815, 2019-11-05:
> Does this only work on SERIAL4 or SERIAL6? I was using SERIAL1.

SERIAL 1 or 6 is correct, I corrected it up.

> Should the OPTION be 4 or 7?

Take OPTION=7, it's always right. OPTION=4 also works because only TX is used. Please install the latest Mission Planner version and flash "Latest" firmware, then you will see in the Full Parameter List also what is meant by the options exactly.

> It's very difficult to find accurate, up to date info on this issue.

The info here is correct.

dposea, 2019-11-05:
Is there any need to add a diode between the TX and RX pins? I tried soldering a wire between TX6 and the small P pad on an R-XSR . I finally got some debug messages i had added to FrskyPassthruTelemetry.cpp, but my X7 still finds no sensors except RSSI and RxBat. It may be possible my radio configuration is not correct because on the telemetry screen i have only default and external options under rssi source, but I am using the internal XJT module. I opened an issue on OpenTX for that.

Target0815, 2019-11-05:
> Is there any need to add a diode between the TX and RX pins?

No, because only TX is used.

> I finally got some debug messages i had added to FrskyPassthruTelemetry.cpp, but my X7 still finds no sensors except RSSI and RxBat.

FrSkyPassthruTelemetry.cpp doesn't tell me anything. That only RSSI and RxBat are found is completely correct. The other values are packed in the Passthrough protocol and are only displayed if you use the Yaapu script.

Alternatively you can use protocol 4 if you are only interested in the normal telemetry values.

manotroll, 2019-11-05:
> Is there any need to add a diode between the TX and RX pins? I tried soldering a wire between TX6 and the small P pad on an R-XSR . I finally got some debug messages i had added to [...]

you could show pictures of how connected and welded R-XSR if possible
I am using the internal module and the latest firmware available for R-XSR
Then it was on SERIAL6 I will test here on this door too

dposea, 2019-11-05:
Thanks very much for your help. Ive been trying for days to get this to work. I added some debug messages to the file that controls passthru. Im running the latest development arduplane, but version 1.8.0 of the yaapu script. In the past I always had to discover sensors, and I still get no telemetry on the yaapu screen. I will look for a newer version.  I will dig a bit more in to the ardupilot code to learn more about how the signal inversion and half duplex is implemented. Wait, inversion is not implemented on the F405, only the F765. Half duplex is implemented, and that just uses the TX pin for send and receive. TX6 is a USART not a UART, that must be why not all serial ports will work. Its beginning to make sense. I will post again if i ever get it to work.

Target0815, 2019-11-05:
> but version 1.8.0 of the yaapu script.

I also use this version, so it works for sure.

The Yaapu script depends on the RSSI value being transmitted. Do you have a value greater than 0 in the normal telemetry screen?

@yaapu Alex can certainly say something about that ...

dposea, 2019-11-05:
Yes, I get RSSI of 90+ if the plane is close. Im not sure why I only get External as an RSSI source. I made sure the R-xsr is bound in d16 mode, with telemetry on channels 9-16. I should also get an oscope in the mail today. Im tired of trying to diagnose communication issue with no way to know what the signal levels are. I would never have tried serial6 if you had not told me. I have GPS on serial4, sbus on 2, and the rest are not being used at this time. I do need a Sik Radio set.  Im using the F405 on a convergence, and it is a great FC.

manotroll, 2019-11-05:
> Yes, I get RSSI of 90+ if the plane is close. Im not sure why I only get External as an RSSI source. I made sure the R-xsr is bound in d16 mode, with telemetry on channels 9-16. I should also get an [...]

but the F405-STD has no serial 6 where and physically serial 6?

Target0815, 2019-11-06:
> I made sure the R-xsr is bound in d16 mode, with telemetry on channels 9-16.

?, why channel 9-16? Please connect the R-XSR to channel 1-8 with telemetry.

> but the F405-STD has no serial 6 where and physically serial 6?

This is about S.Port telemetry on an F405-Wing. If you are using another FC, you have to do the research yourself.

dposea, 2019-11-07:
There are options to bind 1-8 with telemetry, and 9-16 with telemetry. I'll try the 1-8 option. I thought this worked by sending signals from the sticks etc. on 1-8, and telemetry data on 9-16. I could be wrong.

Target0815, 2019-11-07:
@dposea No, that's not right. But it doesn't interest here either, because you probably connected your receiver to the F405-Wing via SBUS. There are always 16 channels transmitted.

Since there is probably a simple basic problem here, please describe/draw/photograph your setup, including the settings in the mixer and attach your ArduPlane parameter file here.

dposea, 2019-11-07:
There is something really wrong. I added debug statements in the code, and data is being sent, but only reciever requests are visible on the TX port. I can post my params file,  and take pics of the wiring, but i have checked those for days. I think im going to look at the UART code and add debug info there. The data should be visible on the tx line.

Target0815, 2019-11-07:
Don't make it so complicated ... The code is fine, but your configuration is not. Please show the desired data, then I can definitely help you, otherwise I am out ...

dposea, 2019-11-08:
I was very busy and did not have time to attach files earlier. I did not mean to be rude or imply I can't be mistaken :)  
[dgp.zip](https://github.com/ArduPilot/ardupilot/files/3822568/dgp.zip)

this is the params file, I'll upload images from my phone.

dposea, 2019-11-08:
[image]
[image]

I think you can tell where the necessary wiring is going. S.port is on TX4, and SBUS is on the SBUS row. I can also post pics of the MAX232 converters I built, but someone said they were not great anyway.

Target0815, 2019-11-08:
With SERIAL6 you can find this in the parameters:

> 1	1	SERIAL6_BAUD	38	6
> 1	1	SERIAL6_OPTIONS	0	4
> 1	1	SERIAL6_PROTOCOL	5

This is wrong, see above ...

Besides, your parameter file looks different than one you can export in Mission Planner. Please flash a "Lastest" version of the firmware from firmware.ardupilot.org/plane. With that (and the right settings) it will work.

dposea, 2019-11-08:
It was from QGroundControl, and I had moved the port to Serial4. Sorry, I meant to move it to Serial1 instead. I'll export from Mission Planner instead. I moved GPS to Serial6 because it's not fully supported by qgroundcontrol (no drop down menus only raw values). I got confused about the ports supporting half duplex, so I will try it on 1.

Target0815, 2019-11-08:
> I got confused about the ports supporting half duplex, so I will try it on 1.

Why don't you just use the UARTs as they are defined for the F405 wing?
SERIAL1 -> Mavlink data
SERIAL2 -> SBUS/PPM Input
SERIAL3 -> first GPS
SERIAL4-> second GPS
SERIAL5-> Mavlink-Data
SERIAL6-> Mavlink data

For S.Port you take SERIAL1 or SERIAL6 with the above settings.

This is easier than reinventing the wheel ...

dposea, 2019-12-07:
I have tried everything suggested with nothing working. Uninverted s.port from r-xsr P pad to TX1, SERIAL1_OPTIONS=4 and 7. Attched are oscilliscope images showing the signals.  By experimentation I've found the signal transmitted from the FC  does not drop to 0 volts, only 3.3V.
[image]

cheshira, 2020-03-11:
Hello, mates! 
I am looking for a solution to the very similar problem! I can't get telemetry working with r9mini connected via hacked s.port to the Matek-F405-wing with Ardupilot v4.0.5 onboard.
I tried to connect to the TX6 pin directly as well as to both X6 pins with a diode. I tried general diode as well as the fast shottki diode.
Here are the settings I use:
SERIAL6_PROTOCOL=4 (also tried 10 for Passthrough. Question here: with his protocol do I need to wait for sensors to be discovered on the Taranis telemetry page? Or Yaapu script captures them directly not nesesarily to be discovered by the Taranis?)
SERIAL6_BAUD=57  (Although baud rate does not matter for SmartPort telemetry, cause it is always a hardcoded value of 57,600bps in ArduPilot, as far as I understand)
SERIAL6_OPTION=4 (tried 7 too, but from the conversation above I figured that 4 or 7 does not matter, since transmission goes through one wire anyway)

Well, tried many combinations, but no success. What to check further - I do not know! Please help!
I rely on experience of the memebers as well as on help from @Target0815 - seems the most knowlegeable ! 
Thank you!

Target0815, 2020-03-11:
Take this converter, it will work with any FrSky receiver at the normal S.Port output: https://www.banggood.com/custlink/vDDGosmMEH

cheshira, 2020-03-11:
@Target0815, thank you! So your bet is that there is some problem with Frsky receivers? Aparently at least with R9Mini/MM and R-XSR? Is it admitted by Frsky, or it is only an assumption at the moment?
Want to hear from community, is my case with R9MM the only recorded case, or there are many more?
Thank you and cheers!

Target0815, 2020-03-11:
I would rather call it a problem of Ardupilot. With iNav and Betaflight it works without problems ... But with the above mentioned converter also with ArduPilot.

ninja-zx11, 2020-03-23:
> Hello, mates!
> I am looking for a solution to the very similar problem! I can't get telemetry working with r9mini connected via hacked s.port to the Matek-F405-wing with Ardupilot v4.0.5 onboard.
> I [...]

S port is working fine on my R9MM with arducopter on omnibusF4 board with these parameters:
`SERIAL1_BAUD 57
SERIAL1_OPTIONS 4
SERIAL1_PROTOCOL 10`

I am using the "Inverted S port" pad on R9MM.

So it should also work on your F4 board too.

[6 more replies]