ArduPilot/ardupilot #12348: Auto Launch without the need for a ground station (similar to iNav)

tenten8401, 2019-09-19:
### Feature request

**Is your feature request related to a problem? Please describe.**
Currently if I want to launch with flight controller assistance I need to put it in stability mode, hold the elevator up and throw it at the same time. Either this or pull out my laptop, connect up with my mavlink dongle, set a waypoint for takeoff, upload said waypoint, flip to auto mode and then throw it. This is a pain as I'd mostly like to use my wing for FPV flying and don't necessarily a ground station or have a person there to help me launch.

**Describe the solution you'd like**
I would like the ability to use auto launch on a switch, similar to how iNav does it here: https://youtu.be/LfoYep7a8jQ?t=65
This would make it much simpler to get the plane in the air without any fiddling around required. This could also possibly be chained with RTL once it reaches a configurable altitude, so it'd launch on it's own, come home and circle above waiting for you to take control or change the flight mode.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2019-09-19:
I'm not a plane expert but I think this is probably similar to the [request to add a takeoff flight mode](https://github.com/ArduPilot/ardupilot/issues/2811).

tenten8401, 2019-09-19:
Looks exactly like what I was looking for, must've missed it while looking for potential duplicates. Hopefully maybe we'll see it implemented soon.

closed 2019-09-19 by tenten8401

rmackay9, 2019-09-19:
@tenten8401, there is a PR here: https://github.com/ArduPilot/ardupilot/pull/11277

I think I've heard @tridge say he plans to get back to this at some point in the not too distant future.

tenten8401, 2019-09-19:
Thanks for the link, looks promising. :)

magicrub, 2020-07-31:
This was added in Plane Release 4.0.0beta4, 19th October 2019