ArduPilot/ardupilot #12350: Issues in BatchLogger

andyp1per, 2019-09-19:
BAT_LGCT cannot possibly be correct as a configuration option. The size is hardcoded to 32 in the logging output and also the buffer size is scaled to a multiple of 32. Setting it to anything greater than 32 will caused free memory reads in the logging code. I have removed this.

    data_write_offset is written and read from different threads but there is no concurrency control. I have made this volatile to try and mitigate, but really there should be locking around the access.

    data_write_offset does not wrap correctly, we basically lose samples while the read offset is wrapping and this is what causes my windowing issues. I have changed the code to wrap correctly.

Fixed in #12349 but not fully tested as yet

**Issue details**

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**

IamPete1, 2022-09-11:
@andyp1per Any updates?

andyp1per, 2022-09-13:
@IamPete1 I need to pick apart my PR to separate these bits. A long way down my priority list.