ArduPilot/ardupilot #12355: Arducopter Version 3.6.10 Using Deprecated Mavlink Message REQUEST_DATA_STREAM

jameslabundis, 2019-09-19:
If there was a previous discussion in the forum, link to it

### Bug report

I'm currently writing software to communicate with a quadcopter (Pixhawk) with Ardupilot as its autopilot. I noticed that I was not receiving any telemetry other than the heartbeat. From here I proceeded to manually request the data streams/messages I needed. In my case I needed to request the GLOBAL_POSITION_INT message.

I assumed that Ardupilot was using the most recent, stable version of Mavlink. I tried using the MAV_CMD_SET_MESSAGE_INTERVAL to request the GLOBAL_POSITION_INT message, but received a MAV_RESULT_UNSUPPORTED message from ardupilot. The code that handles the MAV_CMD_SET_MESSAGE_INTERVAL can be found in GCS_Common.cpp, function name: "handle_command_set_message_interval" and only results in a MAV_RESULT_FAILED or MAV_RESULT_SUCCESS from what I read. Getting a MAV_RESULT_UNSUPPORTED lead me to find that Ardupilot 3.6.10 seems to not support MAV_CMD_SET_MESSAGE_INTERVAL despite having code for it.

I tried using the deprecated REQUEST_DATA_STREAM command to get GPS telemetry, including GLOBAL_POSITION_INT and this worked fine. If the above is true, I would consider Ardupilot using a Mavlink message that was deprecated in 2015 as a bug. Is there anyway this can be updated or is there a reason why the deprecated REQUEST_DATA_STREAM is still being used?

_Please describe the problem_

**Version**
3.6.10

**Platform**
[  ] All
[  ] AntennaTracker
[- ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QuadCopter

**Hardware type**
PixHawk

**Logs**
_Please provide a link to any relevant logs that show the issue_

khancyr, 2019-09-19:
Hello,

This message was set as deprecated for PX4 autopilot but not ArduPilot. 3.6.X series still use it and it works well.
Next version will have support of message interval.

OXINARF, 2019-09-19:
@jameslabundis I'm sorry you faced this issue, but the deprecation was wrong and one-sided as @khancyr explained. We will support the new command in the next major stable releases.

closed 2019-09-19 by OXINARF