ArduPilot/ardupilot #12371: build_iofirmware.py & build_bootloaders.py produce different binaries on master

Jaaaky, 2019-09-21:
#11738 # Bug report

**Issue details**

Tools/scripts/build_bootloaders.py & Tools/scripts/build_iofirmware.py produce different binaries on master than those already-built.
I think an autotest should be added for that, as this is both confusing and risky. The risk is that new IOFirmware or BootLoader get a buggy change on AP_HAL_ChibiOS that takes long to be discovered. It should always be consistent with master. Or always be built as a part of was binary build process.
I simply cannot assume that that those built bootloaders or IO_firmware has been tested by any one.

**Version**
_master_

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2019-09-21:
On Sat, 21 Sep 2019, Jaaaky wrote:

> Tools/scripts/build_bootloaders.py & Tools/scripts/build_iofirmware.py produce different binaries on master than those already-built.

Are you refering to the fact they produce different firmware to the ones 
present in the same commit you're building in?

Firstly, can you spot the catch-22 with the premise they are always 
up-to-date? :-)

We pre-build these binaries to save compile time.  We expect these to be 
periodically updated.

> I think an autotest should be added for that, as this is both confusing and risky. The risk is that new IOFirmware or BootLoader get a buggy change on
> AP_HAL_ChibiOS that takes long to be [...]

They're only committed to the repo (so far only by tridge (and one, once, 
by myself IIRC)) once they've been tested.

What makes you think building these out of build_binaries or the like 
would help on the testing front?  If we *were* to build these as part of 
build_binaries.py and include them in the standard master binaries, and if 
someone did introduce a serious bug into the firmware - then people 
testing master may end up with boards with no workable bootloader on them 
and be up for a bit of work to recover their boards.  Vetting the 
bootloaders makes a bit of sense to avoid this scenario.

Not to say we can't do things differently.

IamPete1, 2021-09-12:
Sounds like were happy with the current process.

closed 2021-09-12 by IamPete1