ArduPilot/ardupilot #12389: Spektrum SRXL parsing of AR8010T fails checksum, misreads throttle channel

dmagree, 2019-09-23:
From email discussions with spektrum, they have confirmed that the AR8010T receiver encodes the 2 bytes of the checksum in reverse order from other receivers. This error has been confirmed in linux and pixhawk boards (all receivers packets fail the checksum).

Also, xplus channels are not parsed according to documentation: https://github.com/SpektrumRC/SpektrumDocumentation/blob/master/Telemetry/Remote%20Receiver%20Interfacing.pdf

**Version**
master branch

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2021-09-12:
@andyp1per this is your area, I wonder if this is still a issue?

andyp1per, 2021-09-13:
SRXL2 is the future - not sure I am that interested in fixing RX specific issues in outdated protocols

IamPete1, 2021-09-13:
@dmagree any update, that certainly sounds like it is just a bug in AR8010T, maybe there is a receiver firmware update from Spektrum?

IamPete1, 2021-09-13:
I found a second report, https://discuss.ardupilot.org/t/srxl-spektrum-ar8010t-support-srxl2/44896?u=iampete

peterbarker, 2024-07-31:
We've completely re-written the parsers since this issue was raised.

Please raise another issue if the problem persists.

closed 2024-07-31 by peterbarker