ArduPilot/ardupilot #12396: AP_RangeFinder: Benewake-TFminiPlus driver should output init failure to ground station

rmackay9, 2019-09-24:
The range finder driver for the Benewake TFMiniPlus checks the version of the lidar and if it's too low, outputs a failure string **to the console**.  Output to the console isn't useful because most users don't have anything connected to the console ([code is here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_Benewake_TFMiniPlus.cpp#L103)).  This leaves the user wondering why it's not working.

We should instead periodically output a text string (aka "send_text") or pre-arm failure message to the user.

tatsuy, 2020-12-21:
This is fixed by #16102 (including #13300).
Please close this if that's OK with you.

peterbarker, 2020-12-21:
Closing, thanks @tatsuy

closed 2020-12-21 by peterbarker