ArduPilot/ardupilot #12398: RC_Channel: auxiliary switch support to set servo output to MIN, TRIM, MAX

rmackay9, 2019-09-25:
We should add new RCx_OPTION values to allow a transmitter's auxiliary switch to set the servo output to the servo's MIN, TRIM or MAX value.  I think this makes sense because we already support controlling Relays from auxiliary switches.

We already allow the SERVOx_FUNCTION to be set to an input channel (i.e. "59" for "RCIN9") which works as a passthrough so the input channel's pwm value is passed through to the output channel.  In very recent versions of AP DO_SET_SERVO commands can also be used on channels setup in this way.

This request is a little different though because the inputs and output ranges can be different and noise on the input channel would not flow through to the output.  This should also work better with DO_SET_SERVO commands.

[This requests comes from this Rover discussion.](https://discuss.ardupilot.org/t/bait-boat-2-servo-command-in-one-wp/46900)

TunaLobster, 2021-02-02:
This could be done with a small script although not all boards have scripting enabled.

I'm assuming both the RCIN and the SERVO should be split into equal thirds? That would take out any small noise on the input signal.

peterbarker, 2021-02-02:
@TunaLobster huh.  I didn't recall this issue existed.  Despite having labelled it.

I actually implemented this one already :-)

https://github.com/ArduPilot/ardupilot/pull/15484/commits

closed 2021-02-02 by peterbarker