ArduPilot/ardupilot #12405: Cubes: .apj should change default parameters so HERE2 via CAN is enabled.

Naterater, 2019-09-25:
### Bug report

**Issue details**

The Here2 is now wired standard as CAN. The default parameters for the firmware should reflect that hardware change. This should help with the following discussion: https://discuss.cubepilot.org/t/here-2-can-update/471

CAN_D1_PROTOCOL：1
CAN_D2_PROTOCOL：1
CAN_P1_DRIVER：1
CAN_P2_DRIVER：1
GPS_TYPE：9
NTF_LED_TYPES：231

**Version**
Plane 4.0.0 beta 2 & others

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Cube black/blue and Here2 GPS

**Logs**
N/A

proficnc, 2019-09-25:
Agreed to a point
My only reservations are that the Here2 is not the majority, so therefore many people will still need to have the serial option.

Better would be a simpler interface on mission planner to help here.

Naterater, 2019-09-25:
Could GPS_Type = AUTO work on CAN GPS's? So far it doesn't work like that.

WickedShell, 2019-09-25:
Yeah if you swapped the default GPS type you break all the existing setups that are currently working. Enabling CAN by default is probably safe, but shifting the GPS type to be CAN would break stuff. This is probably best dealt with on the GCS as @proficnc suggested.

IamPete1, 2021-08-29:
Seems like conclusion is that we cant just just change the defaults because that will break the setups of those using the defaults when they upgrade. In which case I think we might as well close.

closed 2021-08-29 by IamPete1