ArduPilot/ardupilot #12413: Code size blowout

peterbarker, 2019-09-26:
### Bug report

**Issue details**

We've blown the code size out enough that sparky2 is no longer compiling.

The relevant commits *really* shouldn't have led to this blow-out.

**Version**

```
9b746b89db bin/arduplane  975968  1272  129976  1107216
0041874826 bin/arduplane  980544  1272  129976  1111792
96c6544997 BAD
255981c60c BAD
```

Failure on 96c6544997:
```
/home/pbarker/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: bin/arduplane section `.text' will not fit in region `flash'
/home/pbarker/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 4452 bytes
collect2: error: ld returned 1 exit status
```

From git log:
```
96c654499796d0cd3fc6d5121f69eff91db8de7b AP_NavEKF3: moved intermediate variable
s to common memory
00418748268177e85a16a84572b12d9c1c8fef32 AP_NavEKF2: moved intermediate variable
s to common memory
9b746b89dbe49637b09edfccb9c902ad78671809 AP_Math: added fill_nanf()
```

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
sparky2

magicrub, 2019-09-27:
we're back in the supporting-smaller-mem-footprint circus. Perhaps we need a new HAL_BAREBONES_FEATURES that strips a lot more out and then the boards can opt in to more things individually

WickedShell, 2019-09-27:
@magicrub sparky2 isn't a minimized board. Turning that on handily fixes it but removes features it used to have. But at the same time it was a 1M board (just no IO firmware).

tridge, 2019-09-27:
the simplest fix is to remove the forced optimisation of EK2 and/or EK3 to -O3. That saves us about 40k of flash. The -O3 is very expensive for EKF code.
I will do some performance testing and see how much the optimisation actually helps on the F4
hmm, using -O2 instead of -O3 actually saves a lot too ...

closed 2019-09-27 by tridge

lukezhqin, 2020-04-23:
amazing works !

PR #12434 (2019-09-27): Change optimisation from -O3 to -O2
This changes optimisation of math intensive code from -O3 to -O2. A test on a F427 of the EKF2 code shows this is actually slightly faster (although the difference is very small, less than 1 percent).
This lowers flash usage on fmuv2 by about 13k
fixes #12413