ArduPilot/ardupilot #12436: AP_Proximity: improve filtering for 360lidar including SF40c

rmackay9, 2019-09-27:
The SF40c driver (and probably RPLidarA2 and MAV drivers) use the shortest distance when calculating the distance for a "sector" (a sector is normally 45deg).  These sectors are then used in the low level AC_Avoidance library to stop the vehicle before it hits things.  These sectors are not used for the higher level path planning avoidance (aka BendyRuler).

We should increase the filtering applied to the sensor readings so that a single short reading is not enough to trigger a drastic slow-down of the vehicle.  We could use a mode filter or something like the glitch protection used for downward facing lidar.

peterbarker, 2020-01-07:
Related: https://github.com/ArduPilot/ardupilot/issues/12919

@rmackay9 I'm pretty sure you've played around extensively with this recently.  Has this been solved?

rmackay9, 2020-01-07:
@peterbarker, it's not really fixed although I'm also not sure that it's a real problem.  I haven't seen any short readings in my logs.

GlidLov, 2020-05-11:
hi, i'm using the DF40/C and facing this problem also (among others). Stop from obstacle should be WAY more filtered. Now it's like a ripple over the thing...

IamPete1, 2021-09-12:
We now have filtering, https://github.com/ArduPilot/ardupilot/pull/16906

closed 2021-09-12 by IamPete1