ArduPilot/ardupilot #12437: AP_RangeFinder: implement glitch protection in drivers

rmackay9, 2019-09-28:
Copter implements glitch protection up in the sensors.cpp file.  Performance could be improved if this was moved to the AP_RangeFinder library at the moment when the sensor driver is calculating the average of all readings received since the last update() call.

The simplest way to implement this would be to use a mode filter with 3 or perhaps 5 elements.  Every new reading would be pushed into the filter and the range returned would be the output of this filter.  We may or may not want to expose a parameter to allow the user to adjust the mode filter size.

Another alternative could be to implement something closer to Copter's existing glitch filter which ignores changes of over 2m until 2 or 3 readings at this new distance have been received.  If done this way we would probably want to add a new RNGFNDx_GLITCH_DIST or _GLITCH_CM to allow the user to define the glitch size.

WickedShell, 2019-09-28:
This should probably be something drivers can opt out of. For example the WASP device is actually capable of (and with our default parameters is) doing internal filtering already at a higher update rate then ArduPilot gets fed data due to the general low polling rate of the update() call in our vehicles.

peterbarker, 2020-06-01:
Another note is @Hwurzburg  anti-glitching in the HC-SR04 - which is actually just as @rmackay9 described it above (albeit with different constants)