ArduPilot/ardupilot #12449: Copter/Mavlink: Copter doesn't stop moving in Z after stopping sending messages through Mavlink

jorge-san-h, 2019-09-30:
While controlling a copter using Mavros I encountered a problem which seems to be related with Mavlink and even though I encountered it using Mavros I believe it belongs to this forum. I was using the ROS topic **/mavros/setpoint_raw/local** to send velocity commands to Ardupilot with the coordinate frame: **FRAME_BODY_NED**, the type mask: **1984** and the mode: **GUIDED**. The problem appeared when sending a **single message of velocity in Z**. If a single message is sent to X or Y the copter will start moving but stops soon after. However, if a single message is sent to Z, **the copter will not stop moving until told otherwise**. No matter what, if the copter remains in GUIDED mode **it will not stop moving until it receives a specific message of velocity zero in Z**. 

**Version**
ArduCopter V3.7.0-dev

Rajeta-Sarkar, 2021-03-28:
Hi! I'm Rajeta Sarkar and I want to work to fix this issue. What are the Pre requisites and where should I start from?

rmackay9, 2021-03-28:
@Rajeta-Sarkar,

Thanks for looking at this.  The first thing I would do is try to reproduce the issue.  I don't think you'll need to setup ROS but instead can just use regular SITL and send velocity commands using the "velocity" command in MAVProxy.

This issue was raised quite a while ago but i'm a little surprised that I marked it as a bug because from a quick look at the code I don't see the issue.

Rajeta-Sarkar, 2021-03-29:
Actually, I'm an absolute beginner and this is my very first time working on an open-source.
How should I reproduce an issue? 
It would be great if you can guide me with this.

hendjoshsr71, 2021-09-15:
From both code inspection and all of the latest updates to guided with timeouts for vel, acc, and yaw rate commands I do not think this is an issue anymore. 

I am closing this one as I can't replicate it any more. I played with this issue using,
`mode guided`
`arm throttle`
`takeoff 50`
`module load message`
`message SET_POSITION_TARGET_LOCAL_NED 0 1 0 8 1984 0 0 0 1 1 0 0 0 0 0 0` 
With velocities above switched around to 0 & -1 for each axis

If you think it is still an issue in 4.1 please re-open and let us know.

closed 2021-09-15 by hendjoshsr71