ArduPilot/ardupilot #12456: Rename "Bad AHRS" on boot

Naterater, 2019-09-30:
### Bug report

**Issue details**

All vehicles are displaying "**Bad AHRS**" on boot, and it makes new users think that there is an error when in reality the AHRS is OK, just still initializing, usually to get good GPS position.

I propose that the initial status text message sent to the GCS is "AHRS initializing." After the AHRS is healthy, then "Bad AHRS" could be displayed for future bad AHRS events. It could even get more specific, like the following messages:

AHRS aligning
AHRS waiting for GPS lock
AHRS ready

This would help people a lot when setting up vehicles.

This is the face of ardupilot, and we should make it look good. "Bad AHRS" on every boot is ugly.

**Version**
Most recent releases in the last year.
**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
All

**Logs**
N/A

magicrub, 2019-09-30:
I *strongly* agree with this.

WickedShell, 2019-09-30:
This is being driven by the SYS_STATUS bits, which can't convey the extra information. Unless you add an additional message I don't see how to meaningfully get this on a GCS. (I guess I also didn't think it was that bad a message since it was accurate, but I can see how it would be confusing).

Naterater, 2019-09-30:
Whatever has to be done to make it change, I think it should be done. What's the problem with an additional message? Are we really going to run out of messages? The face of ardupilot deserves a few meaningful messages, and other brands of UAS have rather descriptive messages that don't make you think something is potentially wrong or configured incorrectly on every boot.

proficnc, 2019-09-30:
Or change it to “AHRS is fine tuning”
This “error” has been the cause of many hardware returns...

Naterater, 2019-09-30:
I like the suggestions, so I’ll build on them:

AHRS gaining precision
AHRS alignment in progress
AHRS waiting on GPS
AHRS not ready

proficnc, 2019-09-30:
You forgot...
“AHRS message to confuse users”

rmackay9, 2019-09-30:
I totally agree with this as well although this only affects Plane and Rover.  There are two approaches:

- change mission planner because it's not actually the flight code that's sending this message ([Mission Planner issues list is here](https://github.com/ArduPilot/MissionPlanner/issues))
- change the flight code's initialisation so it doesn't set the system-status's ahrs bit until some reasonable time after boot.  This is what Copter does.

WickedShell, 2019-09-30:
My problem with these two approaches are:

>change mission planner because it's not actually the flight code that's sending this message

You get the same message sent in flight when it does go unhealthy, and it's not just waiting on the GPS fix anymore.

> change the flight code's initialisation so it doesn't set the system-status's ahrs bit until some reasonable time after boot. This is what Copter does.

I actually like that the flag is there, as it allows you to see that the vehicle isn't close to being ready to arm itself.

I'd actually suggest we could add bits in the EKF_STATUS_REPORT to indicate it's waiting on the GPS, then the GCS could simply show this message as a higher priority if it's set then the existing bad AHRS.

rmackay9, 2019-10-01:
@WickedShell, I agree with your points especially the first one.  I think the second approach is also probably the best but seems like more work than is required to make users happy.

At least with Copter and Rover's pre-arm checks (I'm less sure about Plane) there are numerous warnings to the user about pre-arm issues because it runs the pre-arm checks every 15sec while disarmed.  .. so I don't think the additional "Bad AHRS" helps users much.  You could even argue that using the ekf-status-report as a way to report pre-arm issues is breaking the logic up (i.e. we would be using send-text and the ekf-status-report to convey pre-arm info to the user).

Anyway, this is really a Plane issue.. for Rover I'll probably just make it act like Copter because it's sufficient to make users happy.

WickedShell, 2019-10-01:
Well I actually really dislike that copter spams that, and am very very glad plane doesn't. We already send the EKF_STATUS_REPORT quite often and it already has flags to say what it's fusing, a flag saying waiting to initialize actually is quite logical extension to it.

rmackay9, 2019-10-01:
I agree that the spamming is an inelegant way to transfer the pre-arm info to the user/gcs but the ekf-status and system-status messages aren't a good solution either for pre-arm checks - they don't have nearly the level of detail required.  I think we have an item on the to-do list to add a special pre-arm check message although I can't find it immediately.

Naterater, 2021-07-26:
I believe this is closed by https://github.com/ArduPilot/mavlink/pull/128

closed 2021-07-26 by Naterater