ArduPilot/ardupilot #12457: EKF2 & EKF3 wind estimate only while maneuvering

Jaaaky, 2019-09-30:
### Feature request
I've done many flight tests to analyze some EKF behavior on plane. One problem I've found is with and without ARSPD_USE wind estimate always gets very bad when plane flies straight forward, it works only when when doing loiter or any maneuvering turns. A forward flight of 2 kilometers causes too much change. 
This is a recurrent always found behavior. I've tested this with and without airspeed on both EKF2 and EKF3.

So like EKF MAG Calibration I think there should be a way to do wind estimated only on maneuvering. And do not change wind estimate when moving forward or almost forward.
Actually the part that is massively affected by wrong wind estimate is dead-reckoning, but also deciding to turn into wind or other parts too.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

magicrub, 2019-09-30:
yeah, thats a good idea. Better to gate the current computed result if we know the "input data" will likely cause it to degrade.

Jaaaky, 2019-10-01:
@priseborough What do you think on this?

samuelctabor, 2019-10-16:
It should be possible to accurately estimate wind speed in purely forward flight too, using difference between airspeed/groundspeed and heading/GPS ground course.

pompecukor, 2020-10-02:
We are getting a lot of this on patrol missions. it is really bad how far the estimation would drift after just a few km.

fredowski, 2023-06-24:
I have problems during takeoff and I also think that this might be due to bad wind and airspeed estimation in the early phase of the takeoff. I have described the behaviour here: https://discuss.ardupilot.org/t/plane-bad-ekf3-wind-and-airspeed-estimation-during-takeoff-with-synthetic-airspeed-takeoff-problems/102797

moreba1, 2025-01-30:
Hi
does this problem of wind estimation was solved ? 
@Jaaaky