ArduPilot/ardupilot #12494: Cube Orange: BLHeli Pass-through unsuccessful

Naterater, 2019-10-04:
### Bug report

**Issue details**

Someone please confirm.... my Cube Orange will not connect to the ESC's in BLHeliSuite32. Everything works exactly OK with my cube black with the same parameters. I get an error that BLHeliSuite cannot connect when I read setup. It does successfully connect to the flight controller and recognize that there should be 2 ESC's available... it just can't communicate with them. Will post screenshots momentarily.

**Version**
Plane 4.0.0 beta 2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin electric

**Hardware type**
Cube Orange, mini carrier board.

**Logs**
N/A

vierfuffzig, 2019-10-04:
@nathan i‘ve been tinkering quite a bit with blheli recently and i think i can say there‘s no fundamental issue in terms of blheli code. 
could you post your parameters?

Naterater, 2019-10-04:
I think I may have had an issue with a carrier board; the telem1 didn't power, and I'll report back with results in a few hours. Sorry if this is a false alarm and really a hardware issue.

Naterater, 2019-11-08:
BlHeli pass-through has been successful with other cube oranges now. Must have been an issue with hardware.

Naterater, 2019-12-09:
Unfortunately as of today it is not working again for me. Maybe it's the BlHeli32's new firmware 32.7. Maybe it's something else, but I can get things to work on the cube blue with the same parameters and firmware.
[image]

[image]

IamPete1, 2020-02-05:
I have tested this, and also the discussion [here ](https://discuss.ardupilot.org/t/orange-cube-4-0-and-blh-passthrough/51296?u=iampete) confirms its not working.

A useful way to test this is to set SERVO_BLH_DEBUG = 1 and SERVO_BLH_TEST to your BLHeli Motors. This triggers a internal ArduPilot test of BLHeli functionality.

Testing with a logic analyzer shows the output has bit framing errors. This is either due to a error in the baudrate, it should be 19200 but is closer to 16500. Or a issue due to protection circuity as was recently solved for serial ports.

[image]

https://github.com/ArduPilot/ardupilot/issues/12547

peterbarker, 2020-02-05:
On Tue, 4 Feb 2020, Peter Hall wrote:

> Testing with a logic analyzer shows the output has bit framing errors. This is either due to a error in the baudrate, it should be 19200 but is closer to
> 16500. Or a issue due to protection circuity as was recently solved for serial ports.

tridge put a fix in for neopixel LEDs for the same problem.

commit 4636a4c458f13ef4b3f3762b4cdbbbd06cf4765b
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Mon Nov 4 13:52:21 2019 +1100

     HAL_ChibiOS: support LEDs with a wider range of frequencies

     this fixed LEDs on FMUv5 boards on first 4 aux channels. We need to
     round up not round down in the resulting bitrate

Might be a matter of changing a "false" to a "true" somewhere.

Peter

IamPete1, 2020-02-05:
@peterbarker Thanks for the tip, I will try this tomorrow if no one beats me to it

Naterater, 2020-10-06:
As of plane 4.0.5 and 4.0.6 beta 1, this still does not work. We receive an unrecognized firmware error on ESC1. Everthing the exact same with a cube black works. We probably spent 4 hours confirming this with different hardware (cube oranges, blacks, ESC's, parameters, etc.). Parameters as follows: https://1drv.ms/u/s!AunBK4rE1ZDCgvZxgrv0KHxQNWegLA?e=AbudOO

IamPete1, 2020-10-06:
@Naterater Does it work on master?

WickedShell, 2020-10-06:
Works on stable copter with a cube orange, I haven't tested plane in the same setup, although as a note for it to work you will have had to either disabled or pressed the safety switch such that surfaces are allowed to move.

IamPete1, 2021-07-26:
I think this has now been resolved.

closed 2021-07-26 by IamPete1