ArduPilot/ardupilot #12498: Broken Link between RCin and Receiver is not detect

ljalbert582, 2019-10-04:
During the vehicle is running properly with transmitter, if I remove the cable between RcIN and receiver. All the input stay in the last value before disconnecting.  Is it possible to detect this situation? Or the hardware does not permit that.

I monitored directly the RcIN input with function:
- hal.rcin->num_channels()
- hal.rcin->read(0)
The value stayed the same before and after disconnection of cable between RcIN and the receiver.

ArduRover V4.0.0-dev (8709b313)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Hardware type**
Pixhawk Cube
Scherrer receiver with PPM output.

[image]

auturgy, 2019-10-04:
There’s likely a setting on your receiver to match the ArduPilot rc failsafe (outputs to zero). What type of receiver is that?

Regards,

James

rmackay9, 2019-10-04:
This looks like a support issue so it should be raised in the support forums here: https://discuss.ardupilot.org/c/ardurover/rover-4-0

I'll close this issue if that's ok.  Sorry to be strict but otherwise our to-do list becomes a support forum.

closed 2019-10-04 by rmackay9

peterbarker, 2019-10-04:
On Fri, 4 Oct 2019, Luc Jalbert wrote:

> During the vehicle is running properly with transmitter, if I remove the cable between RcIN and receiver. All the input stay in the last value before
> disconnecting. Is it possible to detect this [...]

That's expected.  Look around for last_valid_rc_ms - and the way the radio 
failsafes work.

ljalbert582, 2019-10-05:
Thank you guys, sorry for publishing in the wrong place. I had tried on the Ardupilot blog, but I had not had any answers to this situation. Thanks Peter, I'm going to try this way.