ArduPilot/ardupilot #12512: LIS3MDL, AK09916 and IST8310 drivers should not be disabled in HAL_MINIMIZE_FEATURES

andyp1per, 2019-10-07:
### Bug report

The LIS3MDL, AK09916 and IST8310 drivers are disabled by HAL_MINIMIZE_FEATURES
This means that on Kakute boards you can't use many of the mRo devices. This seems like a mistake to me.

**Version**
3.6 and 4.0

**Platform**
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Kakute F7, F4

**Logs**
N/A

tridge, 2019-10-07:
unfortunately just disabling HAL_MINIMIZE_FEATURES fakes KakuteF7 for plane overflow by 8036 bytes. It also doesn't fit for copter.
Those 3 compass drivers cost about 5.5k of flash. That would fit, but it means we'd be quite a bit closer to overflowing again, so it makes development harder

andyp1per, 2019-10-07:
@tridge at some level it doesn't bother me as I can compile up whatever I want, but as these boards get more popular it's going to cause some head-scratching - really took me some time to work out why the compass was not working. If I look at the mRo website all of their compass/gps combinations are LIS3MDL and IST8310

andyp1per, 2019-11-28:
@tridge what about just enabling the LIS3MDL - that seems to be the one everyone cares about?

closed 2021-02-17 by andyp1per