ArduPilot/ardupilot #12513: Docker SITL broken

rkoptev, 2019-10-07:
### Bug report

**Issue details**

I want to run SITL software in Docker container, following the [official wiki](http://ardupilot.org/dev/docs/building-setup-linux.html#building-setup-linux), I successfully executed next commands:
```
git clone https://github.com/ArduPilot/ardupilot
cd ardupilot
git submodule update --init --recursive
docker build . -t ardupilot
docker run --rm -it -v `pwd`:/ardupilot ardupilot:latest bash
```

Now, I'm trying to launch SITL:
```
ardupilot@dfd22d01027e:~$ cd ArduCopter
ardupilot@dfd22d01027e:~/ArduCopter$ sim_vehicle.py -w
bash: sim_vehicle.py: command not found
```
Which is not surprisingly, because Tools/autotest isn't in PATH:
```
ardupilot@dfd22d01027e:~/ArduCopter$ echo $PATH
/usr/lib/ccache:/ardupilot/Tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

After adding Tools/autotest to PATH or launching it from autotest dir:
```
ardupilot@dfd22d01027e:~$ cd Tools/autotest
ardupilot@dfd22d01027e:~/Tools/autotest$ ./sim_vehicle.py -v ArduCopter -L KSFO
Traceback (most recent call last):
  File "/ardupilot/Tools/autotest/sim_vehicle.py", line 25, in <module>
    from pymavlink import mavextra
ImportError: No module named pymavlink
```
Seems like I expecting some missing PATH errors. Is that because I done something incorrectly or because of broken PATH setup during docker image build (I noticed that PATH contains /Tools directory, but not Tools/autotest directory, where script is located which seemed strange to me). And also, why I'm getting ImportError: No module named pymavlink? As I understand, Docker environment must be fully configured to be able to run SITL, or isn't it? If no, why can't I found any instructions on how to run SITL in Docker?

**Version**
2f936b0d3e7a2c7362da0d142b966620a0da4445

**Platform**
[  ] All
[  ] AntennaTracker
[+] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QUAD

**Hardware type**
Ubuntu 18.04

**Logs**
```

```

peterbarker, 2019-10-07:
On Mon, 7 Oct 2019, Ruslan Koptev wrote:

> getting ImportError: No module named pymavlink? As I understand, Docker environment must be fully configured to be able to run SITL, or isn't it? If no,
> why can't I found any instructions on how to run SITL in Docker?

Please paste the output of Docker bringing up the container and/or logs.

The Dockerfile in ArduPilot's root directory should be executing 
install-prereqs-ubuntu.sh which should bring pymavlink in.

rkoptev, 2019-10-07:
@peterbarker , here it is: https://gist.github.com/rkoptev/b6ecc4afb28b46905cd73a7f5c088187

peterbarker, 2019-10-07:
On Mon, 7 Oct 2019, Ruslan Koptev wrote:

> @peterbarker , here it is: https://gist.github.com/rkoptev/b6ecc4afb28b46905cd73a7f5c088187

I can't help but suspect that the environment has been modified for root 
(so it gets the $PATH entries) and that the -U for pip has installed the 
packages into root's directoryes, and that you're actually logging in as 
ardupilot.

Could you check that theory, please?

jveitchmichaelis, 2019-11-24:
Your theory is correct.

See above PR. You can build the container as normal and then login and run `sim_vehicle.py` as expected. If you get python errors, try without using volumes (e.g. just do `docker run -it ardupilot:latest bash`).

 I've used this locally with:

```
FROM ubuntu:16.04
WORKDIR /ardupilot

RUN useradd -U -d /ardupilot ardupilot && \
    usermod -G users ardupilot

ENV USER=ardupilot

RUN DEBIAN_FRONTEND=noninteractive apt-get update && apt-get install --no-install-recommends -y \
    lsb-release \
    sudo \
    software-properties-common \
    git \
    python-software-properties && \
    apt-get clean && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*

# Passwordless sudo for ardupilot user
RUN echo "ardupilot ALL=(ALL) NOPASSWD:ALL" > /etc/sudoers.d/ardupilot \
    && chmod 0440 /etc/sudoers.d/ardupilot

RUN git clone https://github.com/ArduPilot/ardupilot $(pwd)
RUN cd /ardupilot && git submodule update --init --recursive

RUN chown -R ardupilot:ardupilot /ardupilot

RUN bash -c "Tools/environment_install/install-prereqs-ubuntu.sh -y && apt-get install gcc-arm-none-eabi -y" && \
    apt-get clean && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*

USER ardupilot

ENV CCACHE_MAXSIZE=1G
ENV PATH /usr/lib/ccache:/ardupilot/Tools:${PATH}
ENV PATH /usr/lib/ccache:/ardupilot/Tools/autotest:${PATH}
ENV PATH /ardupilot/.local/bin:$PATH

# Build SITL/Copter
RUN ./waf configure --board sitl
RUN ./waf -j8 copter

ENTRYPOINT cd /ardupilot/ArduCopter && sim_vehicle.py -w --no-rebuild
```

In this case all you do is run the container and it'll spawn a copter. I'm not convinced by the approach of adding the entire directory during the build process (over cloning) especially when the instructions just override it by mounting the current directory as a volume.

rkoptev, 2019-11-28:
@jveitchmichaelis, Thank you for this fix. Your solution works like a charm. Also I very much like your approach of cloning the repo inside the container during build as it's suits my needs better, it's good. But   just imagine ardupilot developers need to rebuild the image every time they make any minor changes to the code. So I think current approach is more suitable. Voting up for your PR and waiting for it to be merged.

jveitchmichaelis, 2019-11-28:
Great, glad I could help! There is also this docker image which looks nice (and supports multiple sitls  https://github.com/radarku/sitl-swarm)

closed 2019-11-29 by peterbarker