ArduPilot/ardupilot #12514: nmea output code should use fixed buffer and fix snprintf issues

tridge, 2019-10-08:
The AP_NMEA_Output code should allocate buffers once for max size of strings, and also needs to fix the use of snprintf for the lng_string to ensure it doesn't end up with an unterminated string in %s for next printf

tridge, 2019-10-08:
ping @OXINARF

OXINARF, 2019-10-08:
@tridge I have some questions:

- for allocating buffers, your idea is that it allocates a buffer with the max possible size and then just re-use it right?
- as for snprintf: how could it be unterminated? snprintf always terminates; it can return an error, which code isn't checking for but that's a different matter (one that we have in a lot of places).

peterbarker, 2020-01-07:
good-first-issue for the "use persistent buffer" aspect, not necessarily the `snprintf` thing

lanzlg, 2022-11-10:
I looked at this and removed the allocation of the strings, the static buffers are now private members of the class ( size=186 bytes).

Is there a reason for **not** moving the buffers to the class?   

The behaviour of the snprintf is ok meaning it never produces unterminated strings. However if there are unexpected input values (eg tm_hour >= 100) then this will result in wrong formated messages like
`$GPGGA,1002650.01,....`
 instead of
`$GPGGA,122649.911,...`

(testet with SITL on Ubuntu)

rmackay9, 2022-11-11:
@lanzlg,

In a general sense, using the dynamic allocation could save memory if the function was never called.  In AP though we try to avoid dynamic allocation where possible because it could fail in flight.  It is better to know before we takeoff if we have enough memory for everything or not.