ArduPilot/ardupilot #12518: AC3.6.10 copter quadricopter

roque-canales, 2019-10-08:
Hello,

Testing the end off battery, I had throttle limit issue starting at log line 225000, instead motor was not at 100% (1920pwm max esc calibrated).
So drone slowly fall down, until I released the load. (servo controled).
Also I check Mot thr max param was set to 1.

I Don't understand this behavior, can you explain to me?

You will find attached log of this flight.

[2019-10-08 18-16-50.zip](https://github.com/ArduPilot/ardupilot/files/3703802/2019-10-08.18-16-50.zip)

rmackay9, 2019-10-09:
Sorry to be strict but support really needs to go into the [support forums](https://discuss.ardupilot.org/c/arducopter/copter-36) or else our to-do list turns into another support forum.

lthall, 2019-10-09:
Please provide a link to the discuss post so we can talk about this further.

RickReeser, 2019-10-09:
Discuss topic: https://discuss.ardupilot.org/t/critical-copter-3-6-10-throttle-limit-issue/47970/4

roque-canales, 2019-10-09:
Hello Randy I understand sorry.

But i think I find real issue.....

https://discuss.ardupilot.org/t/critical-copter-3-6-10-throttle-limit-issue/47970/3

Please if you can test it SITL to confirm it, I think after you will able to make correction...

roque-canales, 2019-10-09:
Explanation was gift in support forum.

Its due to correction to be done in the user guide:
« Setting this to the battery failsafe voltage is a good start. »

lthall, 2019-10-10:
When you have current limiting value set it will limit the throttle to a point that will prevent the battery voltage being pulled below your minimum voltage set in the MOT parameters. Appart form prolonging the battery lifetime, it lets you maximise flight time by limiting the power draw when the battery is getting low and causing a battery failsafe.

Notice in the code that it does not pull the maximum throttle below the hover throttle. The lowest value it can pull maximum throttle is: Hover + (1-Hover)*0.2

Now the problem. This should never reduce the maximum throttle below the hover throttle. It should simply limit your maximum lean angle and ability to climb quickly. So it should never cause the aircraft to descend unless the parameters have been set badly. In particular if your hover throttle is set too low.

So we have good news and bad news.... And great news....

The good news is you have not set the parameters badly.
The bad news is we have a bug.
The great news is I found it :slight_smile:

So the problem is we are not accounting for the increase in hover throttle caused by the battery voltage scaling correction. Because your maximum thrust has dropped by 20% at this point in the flight, when you get the reduction it goes lower than it should.

lthall, 2019-10-10:
https://github.com/ArduPilot/ardupilot/pull/12530

closed 2019-10-15 by lthall