ArduPilot/ardupilot #12542: Significant issues with BLHeli on 4.0

andyp1per, 2019-10-12:
I am seeing a number of issues with BLHeli support in 4.0

### Bug report

 - If I enable BLHeli Passthrough and connect with BLHeli, after I disconnect I start seeing the error " internal error (0x8000)"
- If I enable ESC testing I get
```
APM: ESC: Running test on channel 0
APM: ESC: BL_ConnectEx 0/4 at 0
APM: Initialising APM
APM: ESC: BL_ReadBuf 8 -> 9
APM: ESC: Interface type imARM_BLB
APM: ESC: BL_SendCMDSetAddress 0x7C00
APM: ESC: BL_ReadBuf 256 -> 259
APM: ESC: BL_SendCMDSetAddress 0xEB00
APM: ESC: BL_ReadBuf 16 -> 19
APM: ESC: Prot 5 Good 151338 Bad 0 0 0 0 x0

APM: ESC: Test PASSED
Critical failure
``
 - If I enable ESC telemetry I lose logging.

If I had to guess I would say the BLHeli support is pegging the CPU somehow

**Version**
4.0

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad 3"

**Hardware type**
KakuteF7 Mini

**Logs**

andyp1per, 2019-10-12:
If I turn off auto telemetry in BLHeli I no longer get the "Critical failure" when testing but I do still get "Internal errors (0x8000)"

andyp1per, 2019-10-12:
Part of this seems to be related to DMA on the telemetry pin. If I disable this then things are better, rpm output is still a little suspicious however:

[image]

and the Internal errors is still there on test / passthrough

vierfuffzig, 2019-10-12:
i could reproduce this, doing a BLH_TEST on copter firmware reliably triggers internal error 0x8000. it does this too without setting up or connecting ESC telemetry at all. on a sidenote, BLH_DEBUG prompts also do not work on copter.
there's no comparable issues on plane firmware.

closed 2019-10-31 by tridge

andyp1per, 2019-11-04:
I played around with blheli yesterday with this in and all looked good, so I think this has essentially fixed the problems I was seeing.

PR #12677 (2019-10-31): AP_BLHeli: fixed critical errors caused by BLHeli code
fixes #12542