ArduPilot/ardupilot #12547: Copter: 4.0 issues list

rmackay9, 2019-10-12:
This is a list of issues discovered during Copter-4.0 testing that we should resolve before the official release:

- [ ] BLHeli telemetry causes internal error 0x8000 (https://github.com/ArduPilot/ardupilot/issues/12542)

tridge, 2019-10-28:
fix for #12542 here: https://github.com/ArduPilot/ardupilot/pull/12677

tridge, 2019-10-28:
fix for sparky2 here: https://github.com/ArduPilot/ardupilot/pull/12678

andyp1per, 2019-11-15:
It's not true to say dshot600 isn't working - it's working for me fine, must be hardware specific.

GlidLov, 2020-01-10:
today i tried a large quad with cube 2.1, copter 4.0 stable and a Drotek Sirius F9P + RM3100 compass. 
Same quad had here2 (m8p) and 3.6.9 previously. 
Well there's something weird about compass. although it's correctly calibrated, it show ground mag anomoalies while still on ground (sometimes) and does a "ground mag anomaly, inflight yaw allignment" just after take off. 
Sometime it happened with M8P also, but it was a more consistent and predictable way. The thing is that with the new setup, it does toilet bowl heavily when in loiter, then i revert back to stabilize, back in loiter, and then much less toilet bowl, then again stabilize and loiter, and then it "clicks in" and does a proper loiter. For the rest of the flight it will do OK. Land, cycle power, and again heavy toilet bowl, ground mag anomaly, etc....
I think it's something with the RM3100 driver. Satellites are doing ok, i even reached a personal record of 32 sat / 0.4 hdop fix with that!

rmackay9, 2020-01-11:
@GlidLov, thanks for the report.  Could you raise a new topic in the [Copter-4.0 forum](https://discuss.ardupilot.org/c/arducopter/copter-40) and if possible include an onboard log file.  This will make the analysis and communication re this report much easier, thanks!

GlidLov, 2020-01-13:
> @GlidLov, thanks for the report. Could you raise a new topic in the [Copter-4.0 forum](https://discuss.ardupilot.org/c/arducopter/copter-40) and if possible include an onboard log file? This will make the analysis and communication re this report much easier, thanks!
> 
> P.S. I suspect it's interference on the compass from the motors/ESC/powerwires but it will become clear with an onboard log.

well, i just discovered a potential mistake (still investigating): i suspect i'm not using the RM3100 but the infamous internal compass of the CUBE. This is because i have sirius gps plugged into Telem2 (not working in the GPS2 port) which does not have an I2C port. i will have a look into logs later today.

lukezhqin, 2020-01-27:
I really hope you guys would kindly review this issue #13251  which I've already open this PR #13102 though a long time age.
I hope this wp_nav issue could be fine solved before our offical version 4.0 for Copter released.

anbello, 2020-05-05:
I added a comment on https://github.com/ArduPilot/ardupilot/issues/12991 (EKF is not using RangeFinder when EK2_RNG_USE_HGT > 0, EK2_ALT_SOURCE = 0) I think it may be of interest.

omegaretro, 2020-05-18:
I compiled a 4.0.4 master for the pixhawk1-1m, but the rm3100 is not being listed in the compass_typemask which may have been canceled during the compilation appears the rm3100 being compiled

andyp1per, 2020-09-29:
Acro is broken in 4.0.4 from https://github.com/ArduPilot/ardupilot/commit/7fe700b738902c31a8a67919be3ca18bd2323cfa there is a typo:

g.acro_rp_p = constrain_float(g.acro_rp_p, 0.0f, 1.0f); 
should instead be
g.acro_rp_expo = constrain_float(g.acro_rp_expo, 0.0f, 1.0f);

rmackay9, 2020-09-30:
@andyp1per, thanks for finding this.  I guess not many beta testers fly acro!  FYI @lthall.

amilcarlucas, 2021-01-29:
@rmackay9 can you please add : https://discuss.ardupilot.org/t/downloading-dataflash-logs-doesnt-work-in-with-copter-4-0-5/65620 to the list