ArduPilot/ardupilot #12548: 3.5.7 firmware Not building

Mohit953, 2019-10-13:
#### Issue details
I am not able to build the arducopter firmware version 3.6.1 px4-v3
I am trying it with Cygwin. it is showing 

Build failed
 -> task in 'px4_msg_gen' failed (exit status 1):
        {task 7696491021168: cmake_build_task  -> }
/usr/bin/cmake --build /cygdrive/c/ardupilot/build/px4-v3/modules/PX4Firmware --target msg_gen

#### Platform
[  ] Copter

#### Hardware type
Pixhawk2 board

Please help

Thank you

rmackay9, 2019-10-14:
I can confirm that this is a real issue.  I was also not able to build Copter-3.6.1 for the px4-v3 target using "waf".  I was able to build 3.6.1 for the "fmuv3" target though which is the same but using ChibiOS.

I suspect that 3.6.1 will require building with "make" instead of "waf" and we no longer have documentation on how to build with "make" ([build wiki pages are here](http://ardupilot.org/dev/docs/building-the-code.html)) although the documentation is somewhere within [Github](https://github.com/ArduPilot/ardupilot_wiki).

Mohit953, 2019-10-14:
I have started the discussion on the Gitter channel. Still, I am in search of my solution. If you find out anything please share.

muramura, 2019-10-15:
I built AC3.6.1 on ubuntu.
The result was built.
This may be a problem with Cygwin's waf.

＝＝＝

[521/522] PX4: Copying firmware_nuttx to bin/arducopter
[522/522] PX4: Copying firmware and adding git hashes build/px4-v3/bin/arducopter.px4
Waf: Leaving directory `/home/muramura/work/ardupilot/build/px4-v3'

BUILD SUMMARY
Build directory: /home/muramura/work/ardupilot/build/px4-v3
Target          Text     Data  BSS    Total  
---------------------------------------------
bin/arducopter  1085536  2880  59768  1148184

PX4
The ELF files are pointed by the path in the "Target" column. The .px4 files are in
the same directory of their corresponding ELF files.

You can use the option --upload to upload the firmware to the PX4 board if you
have one connected.

PX4IO
Binary                 Text   Data  BSS   Total
-----------------------------------------------
px4-extra-files/px4io  40072   600  3158  43830

Build commands will be stored in build/px4-v3/compile_commands.json
'copter' finished successfully (1m54.437s)
muramura@Aspire:~/work/ardupilot$

colorfuljune, 2019-12-24:
I was also not able to build Copter-3.6.11 for px4-v2 target using "waf“

Waf: Leaving directory `/home/ardupilot/build/px4-v2'
Build failed
 -> task in 'px4_msg_gen' failed (exit status 1):
        {task 7696561263152: cmake_build_task  -> }
/usr/bin/cmake --build /home/ardupilot/build/px4-v2/modules/PX4Firmware --target msg_gen

peterbarker, 2020-01-07:
`px4-v3` is no longer a valid target for Copter :-)

closed 2020-01-07 by peterbarker

auturgy, 2020-01-07:
It is for 3.6.1.
This could just be a submodules or compiler thing: if checking out 3.6.1 the user would need to also do a submodule update, and use the old compiler.

Regards,

James

Irhtn, 2020-11-27:
try ./waf clean