ArduPilot/ardupilot #12562: Wait until heartbeat packets received to enable GCS failsafe

mwrightE38, 2019-10-15:
### Feature request

When using the GCS failsafe I think it's logical to wait until receiving a heartbeat packet to activate.

On plane servos can be stressed while waiting to establish a connection. With some higher end radios there is a longer boot sequence. If GCS failsafe is RTL and RTL to autoland is being used it's possible for the WP number to be set to the beginning of a landing sequence. If the user doesn't catch the WP change before takeoff it could cause an issue.

**Describe the solution you'd like**

I think it seems logical that the autopilot would be waiting to receive a heartbeat packet before assuming a heartbeat packet has been lost as you really can't lose something you've never had. 

It's easy to get around this with throttle FS by powering on the transmitter prior to power on but you really can't get around it with the GCS failsafe.

**Describe alternatives you've considered**

Can be fixed by commenting out this line of code which I'm really not sure what it's here for. 

https://github.com/ArduPilot/ardupilot/blob/8bfab6d2f2e9a4a1d9a942d092ce3b3afba5b095/ArduPlane/system.cpp#L233

I think the only thing it effects besides GCS failsafe is Advanced failsafes? 

The Advanced failsafe would go off if there was 10 seconds without a heartbeat. Usually it takes longer than 10 seconds to connect even with an RFD or 3dr radio. Is this used because of the failsafe pin option? The only other thing this could possibly be doing is giving the GPS 7 more seconds to get a fix?

In talking with @OXINARF He suggested that this is intended behavior from a copter point of view and I understand that perspective as it follows the way other failsafes work.

He suggested other options could be an arming check. 
This could be checking if current WP is 0 or 1 or a non flying aircraft isn't inside a landing sequence. 

I'd agree with this being a good middle ground option. Stressing servos isn't as detrimental as a quadplane being set to the wrong waypoint due to operator error from a q-mode.

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X ] Plane
[ X] Rover
[  ] Submarine

Pedals2Paddles, 2019-10-15:
It doesn't appear any other vehicle type initializes a fake GCS heartbeat.  All the vehicles including plane seem to look for the last heartbeat time to be 0, and skip the check. But for some reason Plane is initializing the time on boot so it will never be 0.

tridge, 2020-01-14:
discussed on dev call, this could be done as a new FLIGHT_OPTIONS bit to change behaviour

tridge, 2020-01-14:
I've opened a PR for this here:
https://github.com/ArduPilot/ardupilot/pull/13262