ArduPilot/ardupilot #12567: ArduPilot doesn't respect Mavlink version of port when sending MAV_CMD_DO_DIGICAM_CONTROL.

pklapperich, 2019-10-15:
### Bug report

**Issue details**

When running ArduCopter in simulator with a real serial port attached to the vehicle so I can test a camera on serial4/uartE I noticed that that `MAV_CMD_DO_DIGICAM_CONTROL` when sent from either clicking "Trigger Camera" within the GCS or if the autopilot generates this messages due to flying a mission containing `MAV_CMD_DO_SET_CAM_TRIGG_DIST` the messages are sent via Mavlink2 even though the port is set to Mavlink1 and the camera is sending heartbeats as a Gimbal via Mavlink1, the `COMMAND_LONG`.

For testing I'm using a Sentera Double 4k. 

**Steps to reproduce**
1. Start the simulator via:
```
~/ardupilot/ArduCopter $ ../Tools/autotest/sim_vehicle.py -v ArduCopter -f quad -L Pyramid -j $(nproc) -A "--uartE=uart:/dev/ttyUSB1"
```
2. Connect a ground station and configure the port parameters:
```
SERIAL2_BAUD 115200
SERIAL2_OPTIONS 0
SERIAL2_PROTOCOL MAVLink1
SR2_EXTRA1 10hz
SR2_EXTRA3 2hz
SR2_POSITION 4hz
```
3. Restart the simulator using the same command as step 1
4. Using the rx line from a second serial port, watch the traffic coming out of the camera. Notice it's all heartbeats with a Mavlink1 delimiter:
`HEARTBEAT {type : 26, autopilot : 0, base_mode : 0, custom_mode : 0, system_status : 0, mavlink_version : 0}; delim: 0xFE`
5. Using the rx line from a third serial port, watch the traffic coming out of the drone. Notice the messages all have a Mavlink1 delimiter:
```
ATTITUDE {time_boot_ms : 4642939, roll : 0.00841472763568163, pitch : -0.6049841046333313, yaw : 3.134385108947754, rollspeed : -0.0005203436594456434, pitchspeed : 0.011106032878160477, yawspeed : 0.0009611798450350761}; delim: 0xFE
SIMSTATE {roll : 0.010485770180821419, pitch : -0.6031168103218079, yaw : 3.1256189346313477, xacc : -5.167874336242676, yacc : 0.0917649194598198, zacc : -8.3803071975708, xgyro : -0.0013637746451422572, ygyro : 0.007096045184880495, zgyro : 0.0012050365330651402, lat : 468800607, lng : -967801686}; delim: 0xFE
AHRS2 {roll : 0.008754936046898365, pitch : -0.6044777631759644, yaw : 3.1205947399139404, altitude : 0.0, lat : 0, lng : 0}; delim: 0xFE
AHRS3 {roll : 0.00841472763568163, pitch : -0.6049841046333313, yaw : 3.134385108947754, altitude : 397.4499816894531, lat : 468800647, lng : -967801684, v1 : 0.0, v2 : 0.0, v3 : 0.0, v4 : 0.0}; delim: 0xFE
HEARTBEAT {type : 2, autopilot : 3, base_mode : 217, custom_mode : 3, system_status : 4, mavlink_version : 3}; delim: 0xFE
GLOBAL_POSITION_INT {time_boot_ms : 4643039, lat : 468800557, lon : -967801684, alt : 397450, relative_alt : 124993, vx : -1009, vy : 2, vz : -1, hdg : 17959}; delim: 0xFE
LOCAL_POSITION_NED {time_boot_ms : 4643039, x : -1.9576263427734375, y : -0.107635498046875, z : -124.9931411743164, vx : -10.09014892578125, vy : 0.025735659524798393, vz : -0.01863064430654049}; delim: 0xFE
ATTITUDE {time_boot_ms : 4643039, roll : 0.00804208219051361, pitch : -0.6060734987258911, yaw : 3.134498357772827, rollspeed : -0.0034370357170701027, pitchspeed : -0.02397013269364834, yawspeed : 0.0016968990676105022}; delim: 0xFE
SIMSTATE {roll : 0.01009929459542036, pitch : -0.6041584014892578, yaw : 3.1257967948913574, xacc : -5.6842360496521, yacc : 0.052538178861141205, zacc : -7.934842586517334, xgyro : -0.0029758054297417402, ygyro : -0.02649005688726902, zgyro : 0.0020633188541978598, lat : 468800516, lng : -967801686}; delim: 0xFE
AHRS2 {roll : 0.00830044411122799, pitch : -0.6055868864059448, yaw : 3.120851755142212, altitude : 0.0, lat : 0, lng : 0}; delim: 0xFE
AHRS3 {roll : 0.00804208219051361, pitch : -0.6060734987258911, yaw : 3.134498357772827, altitude : 397.4499816894531, lat : 468800557, lng : -967801684, v1 : 0.0, v2 : 0.0, v3 : 0.0, v4 : 0.0}; delim: 0xFE
SYSTEM_TIME {time_unix_usec : 1570831512604762, time_boot_ms : 4643039}; delim: 0xFE
AHRS {omegaIx : -0.0026554116047918797, omegaIy : -0.0026360841002315283, omegaIz : -0.002453599590808153, accel_weight : 0.0, renorm_val : 0.0, error_rp : 0.0016389929223805666, error_yaw : 0.009721661917865276}; delim: 0xFE
HWSTATUS {Vcc : 5000, I2Cerr : 0}; delim: 0xFE

```
6. Upload a mission that contains `MAV_CMD_DO_SET_CAM_TRIGG_DIST`. Watching the commands sent to the camera and observe the `COMMAND_LONG` are sent using a Mavlink2 delimiter while the `CAMERA_FEEDBACK` and other messages are sent using Mavlink1
```
AHRS3 {roll : 0.0051164585165679455, pitch : -0.6225802302360535, yaw : 3.1369504928588867, altitude : 397.4499816894531, lat : 468799545, lng : -967801683, v1 : 0.0, v2 : 0.0, v3 : 0.0, v4 : 0.0}; delim: 0xFE
COMMAND_LONG {'target_system': 0, 'target_component': 0, 'command': 'MAV_CMD_DO_DIGICAM_CONTROL (203)', 'confirmation': 0, 'param1': 0.0, 'param2': 0.0, 'param3': 0.0, 'param4': 0.0, 'param5': 1.0, 'param6': 0.0, 'param7': 0.0}; delim: 0xFD
CAMERA_FEEDBACK {time_usec : 1570831513800000, target_system : 0, cam_idx : 0, img_idx : 104, lat : 468799475, lng : -967801683, alt_msl : 395.8899841308594, alt_rel : 123.43000030517578, roll : 0.29999998211860657, pitch : -35.77000045776367, yaw : 179.72999572753906, foc_len : 0.0, flags : 0, completed_captures : 0}; delim: 0xFE
ATTITUDE {time_boot_ms : 4644239, roll : 0.005234807264059782, pitch : -0.6250613927841187, yaw : 3.1368649005889893, rollspeed : -0.0008950171759352088, pitchspeed : -0.025688666850328445, yawspeed : -0.000348108122125268}; delim: 0xFE
```
7. Use the manual trigger button in the GCS and observe the same behavior as step 6.

**Version**
Tested with ardupilot rev 05d10467

**Platform**
only tested on copter so far.

**Airframe type**
quad

**Hardware type**
SITL

**Logs**
See steps to reproduce above.

rmackay9, 2019-10-16:
Thanks very much for this detailed report.  I've discussed with @peterbarker who is more an expert in this area and I hope we can get to the bottom of it in the near future.

pklapperich, 2019-10-16:
I will say that I haven't tried in simulator on ArduCopter 3.6, but on the drone at least ArduCopter 3.6 does behave correctly.

rmackay9, 2019-10-21:
Peter Barker has created a fix for this issue here: https://github.com/ArduPilot/ardupilot/pull/12576

I plan to test this fix this week.

rmackay9, 2019-10-23:
@pklapperich, do you think you could re-test with this patch?  If it makes it easier I can produce a binary for you especially if you tell me what type of flight controller you are using.

pklapperich, 2019-10-29:
Sorry for the delay, yes I can test later today. I'm using software in the loop.

pklapperich, 2019-10-31:
With the physical camera attached to ArduCopter SITL via a USB serial adapter to uartE things appear to be working fine now. I tested both clicking the "Trigger Camera" button in the ground station as well as flying a mission that included `MAV_CMD_DO_SET_CAM_TRIGG_DIST`. In both cases the camera received the trigger command in a MAVLINK1 frame and snapped a photo.

But then I changed the `SERIAL4_PROTOCOL` parameter from `MAVLink1` to `MAVLINK2`, and the camera still lit up green (indicating it received the `GLOBAL_POSITION_INT` messages it wanted) and also took pictures when I clicked the "Trigger Camera" button in the ground station. This camera device does not understand `MAVLINK2`, so I didn't expect this to work. I'm guessing that the patch is causing ArduPilot to ignore the `SERIAL4_PROTOCOL` parameter and instead communicates based on the heartbeats it's receiving. Is this expected behavior? I'm not complaining; I'm just not sure it was intentional.

When I changed the `SERIAL4_PROTOCOL` parameter to GPS the camera didn't work (as was expected).

peterbarker, 2019-11-04:
@pklapperich We still start off by sending mavlink1 on a mavlink2 channel - if mavlink2 signing is disabled.  If we receive a mavlink2 packet we start to send mavlink2.

So I think that's all good.

closed 2019-11-26 by tridge