ArduPilot/ardupilot #12570: Copter: package place gets stuck if lat,lon are defined

rmackay9, 2019-10-16:
Based on the discussion here it seems that Copter's Package Place mission command gets stuck if the user defines the lat and lon.  A work around is to use a waypoint command to get to the desired location and then use a package-place command with no lat,lon defined but we should still fix the underlying problem.

nkruzan, 2022-01-17:
Looks like maybe this just need to have cmd.content.location.lat = packet.param5 and cmd.content.location.lng = packet.param6 set here https://github.com/ArduPilot/ardupilot/blob/c1c329e2b4f2750deae14d14c95ee87aebd2248b/libraries/AP_Mission/AP_Mission.cpp#L1135

does that seem right? I'll work on a pr if i'm on right track.

rmackay9, 2022-01-18:
@nkruzan

That sound like it could be the issue, txs!

lthall, 2022-07-28:
Could you please check this again using #21296.

Thanks for catching this!!

peterbarker, 2022-07-28:
I am unable to replicate here.  The vehicle drops the payload at the specified coordinates (augmented test here: https://github.com/ArduPilot/ardupilot/pull/21298/files)

There's no link to the original discussion.  Leonard *has* fixed issues where it might get stuck in his PR if the vehicle didn't have a gripper configured (IIRC).

Since there's no link to further information, I think we have to assume it's either been fixed already, or that the original vehicle didn't have a gripper enabled (in which case it wouldn't have worked without lat/lng supplied either.

The code line referenced isn't relevant - the message is marked as "stored in location", meaning we fill lat/lng/alt in a different part of the code.

Please reopen this if more information is/becomes available.

PR #21296 (2022-07-28): Copter: Fix Payload Place Bug
This fixes a bug introduced in the previous patch and 2 problems previously present.

This also fixes #12570
closed 2022-07-28 by peterbarker