ArduPilot/ardupilot #12579: Using NMEA - HDT message as heading/yaw source instead of compass

bigboy061293, 2019-10-17:
Hi, 

I am kindly new to both github discussions and ardupilot/arducopter firmware. I have read all the discussion here but to be honest that I have no idea, already tried but does not help.

https://github.com/ArduPilot/ardupilot/pull/7215

What I am trying to do is replacing/lowering the weight the compass yaw angle by heading angle from HDT message (NMEA).

I have the GPS output GPGGA and GPHDT. Right now, GPS is read (readable long/lat in Mission planner status) but YAW is still drifting. GPHDT is fine when checked it standalone with terminal software . I disable COMPASS_USE, EK2_YAW_M_NSE, EK3_ENABLE are set, what else parameters do I miss?

[image]

Sorry for duplicating the issue and thank you in advance.

WickedShell, 2019-11-12:
Closing as a support issue, and based on posts on discuss I believe it has otherwise been solved already.

closed 2019-11-12 by WickedShell

rmackay9, 2019-11-13:
@WickedShell, we really need docs on how to set this up though.  It's come upon the Rover 4.0 testing thread as well.  I've added it to our [Rover-4.0 wiki to-do list](https://github.com/ArduPilot/ardupilot_wiki/issues/2109) so if you know how to set it up could you post comments somewhere?

WickedShell, 2019-11-13:
@rmackay9 the only docs I really know of were dropped by tridge in #7215

rmackay9, 2019-11-13:
@WickedShell, aha! thanks.  maybe that's enough..