ArduPilot/ardupilot #12591: Startup error

1623664, 2019-10-19:
`dron@dron-VirtualBox:~/ardupilot/ArduCopter$ sim_vehicle.py --console --map
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/dron/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/dron/ardupilot 
Setting out to                           : /home/dron/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.15 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.15 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.224s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/dron/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/dron/ardupilot/build/sitl'
Embedding file sandbox.lua:libraries/AP_Scripting/scripts/sandbox.lua
Waf: Leaving directory `/home/dron/ardupilot/build/sitl'
Build failed
Traceback (most recent call last):
  File "/home/dron/ardupilot/modules/waf/waflib/Runner.py", line 395, in task_status
    return tsk.runnable_status()
  File "Tools/ardupilotwaf/git_submodule.py", line 75, in runnable_status
    out = self.generator.bld.cmd_and_log(cmd, quiet=Context.BOTH, cwd=self.cwd)
  File "/home/dron/ardupilot/modules/waf/waflib/Context.py", line 461, in cmd_and_log
    raise e
WafError: Command ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/dron/ardupilot/modules/gtest') returned 128

SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks
`

Error on first start. What am I doing wrong?

peterbarker, 2019-10-20:
On Sat, 19 Oct 2019, 1623664 wrote:

> WafError: Command ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/home/dron/ardupilot/modules/gtest') returned 128

Please do a

git pull

then a

git submodule update --init --recursive

ad retry.

1623664, 2019-10-21:
Thank you very much. The error was in the proxy

closed 2019-10-21 by 1623664

iwishiwasaneagle, 2020-06-22:
Hi there,
I have followed your advice but still receive the same error.

```
Setting top to                           : /vagrant 
Setting out to                           : /vagrant/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.17 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.17 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.373s)
Waf: Entering directory `/vagrant/build/sitl'
Command ['/usr/bin/git', 'rev-parse', '--short=8', 'HEAD'] returned 128
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/vagrant/modules/waf/waf-light" "configure" "--board" "sitl"
SIM_VEHICLE: Building
SIM_VEHICLE: "/vagrant/modules/waf/waf-light" "build" "--target" "bin/arducopter"
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks
```

er-jk, 2022-05-06:
Hi @iwishiwasaneagle , Have you found a solution to this problem? When I try to construct SITL, I get the same issue.

```

Build failed
Traceback (most recent call last):
File "ardupilot/modules/waf/waflib/Runner.py", line 395, in task_status
    return tsk.runnable_status()
ardupilot/modules/waf/waflib/Task.py", line 679, in runnable_status
    new_sig = self.signature()
File "Tools/ardupilotwaf/ap_persistent.py", line 32, in _signature
    s = _original_signature(self)
ardupilot/modules/waf/waflib/Task.py", line 847, in sig_implicit_deps
    (bld.node_deps[key], bld.raw_deps[key]) = self.scan()
  File "Tools/ardupilotwaf/mavgen.py", line 32, in scan
    tree = et.parse(node.abspath())
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1182, in parse
    tree.parse(source, parser)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 656, in parse
    parser.feed(data)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1659, in feed
    self._raiseerror(v)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1523, in _raiseerror
    raise err
ParseError: mismatched tag: line 1035, column 6

```

iwishiwasaneagle, 2022-05-09:
@er-jk no sorry. Don't think I ever did

er-jk, 2022-05-09:
> @er-jk no sorry. Don't think I ever did

It's fine @iwishiwasaneagle , clearing the build files solves the problem.

_I clone it after deleting all the build files. but _

Another option is to wipe the wag file from Audopilot like below.(recommended)

```
~/src/ardupilot $ ./waf clean 
          Embedding file locations.txt:Tools/autotest/locations.txt
          Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
          'clean' finished successfully (1.056s)
```

samwidarr, 2024-07-25:
Hello everyone, I'm new, please help me with this, I did everything step by step. Thank you.

add_group: name git_submodules already present

(venv-ardupilot) samwidarr@penguin:~/ardupilot$ ./waf configure --board MatekF405-TE
Setting top to                           : /home/samwidarr/ardupilot 
Setting out to                           : /home/samwidarr/ardupilot/build 
Autoconfiguration                        : enabled 
Checking for program 'python'            : /home/samwidarr/venv-ardupilot/bin/python3 
Checking for python version >= 3.6.9     : 3.11.2 
Setting board to                         : MatekF405-TE 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/arm-none-eabi-gcc 
Checking for program 'arm-none-eabi-nm'  : /opt/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-nm 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 10.2.1 
Checking for program 'make'              : /usr/bin/make 
Checking for program 'arm-none-eabi-objcopy' : /opt/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-objcopy 
Including /home/samwidarr/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/include/minimize_fpv_osd.inc
Including /home/samwidarr/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/include/minimize_common.inc
Removing HAL_MSP_GPS_ENABLED
Including /home/samwidarr/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/include/no_bootloader_DFU.inc
Setup for MCU STM32F405xx
No default parameter file found
Writing hwdef setup in /home/samwidarr/ardupilot/build/MatekF405-TE/hwdef.h
Writing DMA map
Setting up as normal
No change in hwdef.h
Generating ldscript.ld
Checking for env.py
env set ENABLE_DFU_BOOT=0
env set WITH_FATFS=1
env set PROCESS_STACK=0x1C00
env set MAIN_STACK=0x600
env set IOMCU_FW=0
env set PERIPH_FW=0
env set BOARD_FLASH_SIZE=1024
env set EXT_FLASH_SIZE_MB=0
env set INT_FLASH_PRIMARY=False
env set ENABLE_CRASHDUMP=False
env set CPU_FLAGS=['-mcpu=cortex-m4', '-mfpu=fpv4-sp-d16', '-mfloat-abi=hard', '-DARM_MATH_CM4', '-u_printf_float']
env set CORTEX=cortex-m4
env set APJ_BOARD_ID=1054
env set APJ_BOARD_TYPE=STM32F405xx
env set USBID=0x1209/0x5741
env set FLASH_RESERVE_START_KB=64
env set EXT_FLASH_RESERVE_START_KB=0
env set FLASH_TOTAL=983040
env set HAS_EXTERNAL_FLASH_SECTIONS=0
env set CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F4xx/platform.mk MCU=cortex-m4 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1
Enabling ChibiOS asserts                     : no 
Disabling Watchdog                           : no 
Enabling malloc guard                        : no 
Enabling ChibiOS thread statistics           : no 
Enabling -Werror                             : yes 
Checking for intelhex module:                : OK 
Enabled OpenDroneID                          : no 
Enabled firmware ID checking                 : no 
GPS Debug Logging                            : no 
Enabled custom controller                    : no 
Checking for HAVE_CMATH_ISFINITE             : yes 
Checking for HAVE_CMATH_ISINF                : yes 
Checking for HAVE_CMATH_ISNAN                : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
[... 43 lines cut ...]
Checking for env.py
env added ENABLE_DFU_BOOT=0
env added WITH_FATFS=1
env added PROCESS_STACK=0x1C00
env added MAIN_STACK=0x600
env added IOMCU_FW=0
env added PERIPH_FW=0
env added BOARD_FLASH_SIZE=1024
env added EXT_FLASH_SIZE_MB=0
env added INT_FLASH_PRIMARY=False
env added ENABLE_CRASHDUMP=False
env appended CPU_FLAGS=['-mcpu=cortex-m4', '-mfpu=fpv4-sp-d16', '-mfloat-abi=hard', '-DARM_MATH_CM4', '-u_printf_float']
env added CORTEX=cortex-m4
env added APJ_BOARD_ID=1054
env added APJ_BOARD_TYPE=STM32F405xx
env added USBID=0x1209/0x5741
env added FLASH_RESERVE_START_KB=64
env added EXT_FLASH_RESERVE_START_KB=0
env added FLASH_TOTAL=983040
env added HAS_EXTERNAL_FLASH_SECTIONS=0
env added CHIBIOS_BUILD_FLAGS=USE_FATFS=yes CHIBIOS_STARTUP_MK=os/common/startup/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk CHIBIOS_PLATFORM_MK=os/hal/ports/STM32/STM32F4xx/platform.mk MCU=cortex-m4 ENV_UDEFS=-DCHPRINTF_USE_FLOAT=1
Embedding file font0.bin:libraries/AP_OSD/fonts/font0.bin
Embedding file hwdef.dat:/home/samwidarr/ardupilot/build/MatekF405-TE/hw.dat
add_group: name git_submodules already present
(venv-ardupilot) samwidarr@penguin:~/ardupilot$