ArduPilot/ardupilot #12601: AP_Mission: allow missions to be stored on SD cards to allow more points

rmackay9, 2019-10-20:
there have been a few requests from users including in this forum to support very large numbers of waypoints (thousands) which leads to the request to store them on the SD card.

There's a similar request but the justification is to allow easily swapping SD cards to change the missions.

IamPete1, 2020-09-04:
This script example allows loading scripts from SD, could be extended to switch missions with param or rc switch. One could put the both script and mission on a SD card that would then load its mission onto the flight controller.

Of course this is simply loading from the SD, the mission is still run from memory so there is no change to maximum number of waypoints.

https://github.com/ArduPilot/ardupilot/pull/15239

peterbarker, 2024-07-31:
We do this now.

closed 2024-07-31 by peterbarker