ArduPilot/ardupilot #12608: Copter: add target final position to GUID logging

rmackay9, 2019-10-21:


TanmayRajore, 2020-02-05:
I want to work on this issue and contribute to this can you please provide me what file to work on and what exactly you expect the new enhancement to do.

peterbarker, 2020-02-06:
@TanmayRajore First thing - you must get to the point you can run ArduPilot in SITL.

Are you at that point yet?  https://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html

To answer your question, you need to modify this log message:

```
    { LOG_GUIDEDTARGET_MSG, sizeof(log_GuidedTarget),
      "GUID",  "QBffffff",    "TimeUS,Type,pX,pY,pZ,vX,vY,vZ", "s-mmmnnn", "F-000000" },
```

to contain a "final position" target.  When moving around obstacles this may be different to where the vehicle is currently headed.

The relevant code is in Copter/Rover mode_guided.cpp

TanmayRajore, 2020-02-06:
I know how to use ArduPilot in SITL and I will get to work on it and understand it more. Thank you

rishabsingh3003, 2020-04-05:
@rmackay9  Shouldn't this be done for Rover too? Since rover has the GUID message too. I see the PR marked as Copter.

closed 2022-02-10 by rmackay9