ArduPilot/ardupilot #12616: FS_GCS_ENABLE = 1 doesn't work in AUTO

tizianofiorenzani, 2019-10-21:
### Bug report

**Issue details**
Setting the Failsafe parameter FS_GCS_ENABLE to 1 should always set the copter in RTL, but it doesn't work is the Copter is in AUTO.

I have checked the current [master ](https://github.com/ArduPilot/ardupilot/blob/e84082a099e48390c6bf3db9dbda9d7d369c6e0b/ArduCopter/events.cpp#L145) and everything looks fine.

**Version**
Tested in current Copter SITL (3.7.0-dev 57ac582a)

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
SITL

peterbarker, 2020-01-07:
@Pedals2Paddles does it work now?

Pedals2Paddles, 2020-01-07:
Yep. This issue was opened on 10/21/2019, which was before the failsafe refactoring (https://github.com/ArduPilot/ardupilot/pull/7432) was merged on 11/5/2019.  I declare ye closed.

closed 2020-01-07 by Pedals2Paddles