ArduPilot/ardupilot #12618: driver fix for ported chibiOS on stm32f407vet6

adezie, 2019-10-22:
recently I've successfully port chibiOS to stm32f407vet6  [stm32F407](https://os.mbed.com/users/hudakz/code/STM32F407VET6_Hello/shortlog/) but I have a problem that every time I connected the stm32, I have to reinstall the driver so that mission planner could detect the device but for every connection like I2C and SPI doesn't found a problem yet

Azkarel, 2020-05-12:
Dear Adezie.

It looks that you have reach something too much people are trying to do. Install Ardupilot on a stm32f407vet6 with only 0.5mb.
I personally was trying to do the same for the last 6 months unsuccesfully. Other people has trying too:

https://discuss.ardupilot.org/t/porting-to-stm32f407ve-512kb-flash-192-4kb-sram/36903

Can you be so kind to give us more information about this project?

Thank you in advance.

Best Regards

adezie, 2020-05-12:
it will be my pleasure to help you

I've reached it by accident actually, and it worked, I can send the
configuration file for the bootloader and the firmware if you want

Azkarel, 2020-05-24:
Dear Adecie,

Sorry for my answer delay but these are crazy days in the world.

I am still interested in you files. If you are so gentle, please send me this files to my e-mail:
azkarel63@gmail.com.

I am interested in the type of board you are using. I am thinking in use the board attached in the photo. 
[image]

thank you very much in advance.

IamPete1, 2021-09-12:
I don't think we can investigate drivers until support is in tree, or at least a open PR.

closed 2021-09-12 by IamPete1