ArduPilot/ardupilot #12624: Tools: install-prereqs-mac.sh script is broken

jmachuca77, 2019-10-22:
**Issue details**

The script ```install-prereqs-mac.sh``` is broken as it seems PX4 moved its home-brew packages. When running the install script I get the following error:

```
==> Installing genromfs from ardupilot/px4
==> Downloading http://px4-travis.s3.amazonaws.com/toolchain/genromfs-0.5.2.tar.gz

curl: (6) Could not resolve host: px4-travis.s3.amazonaws.com
Error: An exception occurred within a child process:
  DownloadError: Failed to download resource "genromfs"
Download failed: http://px4-travis.s3.amazonaws.com/toolchain/genromfs-0.5.2.tar.gz

==> Installing gcc-arm-none-eabi from ardupilot/px4
==> Downloading http://px4-travis.s3.amazonaws.com/toolchain/gcc-arm-none-eabi-5_4-2016q3-20160926-mac-d.tar.bz2

curl: (6) Could not resolve host: px4-travis.s3.amazonaws.com
Error: An exception occurred within a child process:
  DownloadError: Failed to download resource "gcc-arm-none-eabi"
Download failed: http://px4-travis.s3.amazonaws.com/toolchain/gcc-arm-none-eabi-5_4-2016q3-20160926-mac-d.tar.bz2
```

**Version**
Master

**Platform**
[ x ] All

**Airframe type**
N/A

**Hardware type**
N/A

peterbarker, 2020-01-07:
@lvale I don't suppose you know where these might have gone?

lvale, 2020-01-14:
The links by Jaime are working fine @peterbarker

peterbarker, 2020-01-15:
Looks like @jmachuca77 fixed this.  Please reopen if I'm wrong.

closed 2020-01-15 by peterbarker