ArduPilot/ardupilot #12625: ADSB: _OFFSET_LAT, OFFSET_LON should be renamed to POS_X/Y/Z

rmackay9, 2019-10-23:
ADSB appears to use non-standard position offset parameter names, ADSB_OFFSET_LAT/LON which I think should be renamed to be ADSB_POS_X/Y/Z to be consistent with other sensors ([wiki](http://ardupilot.org/copter/docs/common-sensor-offset-compensation.html))

Maybe I'm misunderstanding but I don't see how a "LAT", "LON" parameter can be a good name for something that is in body frame.

Because users look at parameters directly a lot, I think we need to be careful about how we name things.

rmackay9, 2019-10-23:
@magicrub, this library is perhaps mostly your baby (at least originally) so maybe you have an opinion on this?

peterbarker, 2019-10-24:
On Tue, 22 Oct 2019, Randy Mackay wrote:

> ADSB appears to use non-standard position offset parameter names, ADSB_OFFSET_LAT/LON which I think should be renamed to be ADSB_POS_X/Y/Z to be
> consistent with other sensors (wiki)
>
> Maybe I'm [...]

This is lateral and longitudinal (which are body frame), not latitude and 
longitude (obviously earth frame).

Both should be done with x/y for consistency, as you say.

magicrub, 2021-11-10:
The weird part about this is the terminology is from the ADSB spec. It's obviously not ideal..

shbhmexe, 2025-10-26:
Hi, could you please assign this issue to me? I’ve reviewed the project’s Code of Conduct and Contribution Guidelines, and I have the required skills and understanding of the tech stack to work on this effectively. I’ve already started analyzing the codebase and begun initial work toward resolving the issue. I’m confident I can implement a clean, optimized, and maintainable solution that aligns with the project’s standards. Looking forward to contributing!