ArduPilot/ardupilot #12632: Add parameter to store Mission Hash.

MikeCharikov, 2019-10-23:
### Feature request

Need to add parameter to store mission hash, so GCS can check this parameter before download mission from UAV. No need to implement CRC calculation on autopilot, GCS can calculate this parameter itself and store it on autopilot. 

**Is your feature request related to a problem? Please describe.**

For large mission it takes time to download it all. If GCS knows that uploaded mission is equal to GCS-cached one - a lot of time will be saved in the field.

**Describe the solution you'd like**

To add Int32 parameter to MIS group: MIS_HASH

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

MikeCharikov, 2019-10-23:
At the present time for this task we use parameter that is not used for common flights. Something like VISO_POS_X. 

But it is not the way it should work. :)

auturgy, 2019-10-23:
G’day Mike,
the discussion here https://github.com/mavlink/mavlink/pull/1172 (and the related issue) seems relevant to this. Perhaps you could comment there about your implementation

OXINARF, 2019-10-23:
This is being tracked at https://github.com/mavlink/mavlink/issues/1171 as a generic solution so I'm closing this ArduPilot-specific request.

closed 2019-10-23 by OXINARF

MikeCharikov, 2019-10-23:
Ok.