ArduPilot/ardupilot #12633: Option to remove simple_goto orbiting behaviour for plane/quadplane

LeonFargher, 2019-10-23:
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

I want to use dronekit to create a mission in python, then dynamically command the plane to follow these waypoints, commanding waypoints as it goes along and tracking it's progress.

The problem with this is that to get around the orbiting that arduplane does when it reaches a wp without a wp after it causes me to have to code functionality around it to combat this problem. This causes very unreliable flight dynamics, as well as host of other problems (missing WPs, not landing etc)

Sending MAV_CMD_NAV_WAYPOINT to the pixhawk as a mission allows it to fly through the waypoints and will only orbit at the last waypoint, but the path of the plane cannot be dynamically controlled.

simple_goto allows for dynamic control but will orbit a waypoint until the new wp is commanded.

If its possible, is there a parameter that disables the plane's orbiting behaviour in guided so that the simple_goto sends it straight to the wp with a radius of zero?

**Additional context**
Controlling a very large quadplane that requires stable flight dynamics, given only by the flight control software of ardupilot

Scorpion11k, 2020-07-29:
Any updates regarding sending the MAV_CMD_NAV_WAYPOINT   and receiving (COMMAND_ACK(ID=16,RESULT=3) (UNSUPPORTED) ?

IamPete1, 2021-08-29:
See  Q_GUIDED_MODE

https://ardupilot.org/plane/docs/quadplane-auto-mode.html#guided-mode

closed 2021-08-29 by IamPete1