ArduPilot/ardupilot #12641: Mission loaded on the drone are being repeated after disarm.

tecnic08, 2019-10-24:
### Bug report

**Issue details**
Mission loaded in the EEPROM are being repeated on ground after landing and a failure to get terrain altitude.

I was flying a drone in an auto mission (auto mode) with terrain following.

When my drone completed the mission, landed, and disarmed, the buzzer repeatedly sounded a "waypoint reached" sound and GCS showing "Reached command #n" where n goes 1, 2, 3, 4... (all the NAV_WP command) until my last waypoint. During that, I have a set servo command (for crops spraying), those command are being repeated too. The spray goes on and off repeatedly. All of this happens in AUTO mode. I can repeated this behavior over and over again.
I'm uploading a video of this bug, it'll be in the comment.

I tried again with the same mission, same drone, same take off and landing area, same condition, but without terrain following. The drone completed the mission, landed, and disarm without repeating the mission. No issue or whatsoever.

The drone has a rangefinder, but when it landed, the rangefinder are too close to the ground and it will return an out of range data.

So, I dug deep in the code and found that. This bug lies in `AC_WPNav::set_wp_origin_and_destination`

When the drone disarmed, `AP_Mission::reset()` is called and reset all the command and index to the first command. Then, the first command is loaded into the que by `AP_Mission::update()` that calls `advance_current_nav_cmd`. On the next iteration, `_cmd_verify_fn(_nav_cmd)` is being called to check if that loaded command has been done yet. For Copter, `Copter::ModeAuto::verify_command()` is called.

Some of the command such as `NAV_TAKEOFF` `NAV_WAYPOINT` `NAV_SPLINE_WAYPOINT` etc. is verified by an `AC_WPNav` flag called `reached_destination` and here bug lies here.

`reached_destination` will only be set to false if `AC_WPNav::set_wp_origin_and_destination` (for normal waypoint) successfully runs. Which in my case, does not, because when it landed, the rangefinder are too close to the ground and returning bad data. So, `reached_destination` is not being reset and when `_cmd_verify_fn(_nav_cmd)` in `AP_Mission::update()` tries to verify the command, it always return true if those command use this flag to check. Causing a sequence of waypoint reached sound and other command being done on ground after disarm.

My drone is running a Copter 3.5.7, but the code in Copter 3.6 stills contain this bug.

I'll submit a pull request to fix this problem. The `reached_destination` should be reset as soon as `AC_WPNav::set_wp_destination`  and `AC_WPNav::set_spline_destination` is being called.

**Version**
3.5.7 (Bug still present in 3.6.11, but I haven't tested)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad with 30" props and a laser rangefinder

**Hardware type**
Holybro Pix32

**Logs**
I didn't enable logging after disarm.

tecnic08, 2019-10-24:
I haven't test it yet. If someone can test it, it will be a great help.

peterbarker, 2020-01-07:
@tecnic08 did you end up creating a PR for this - or know that it is fixed in some other way?

tecnic08, 2020-01-07:
@peterbarker Yes, here https://github.com/ArduPilot/ardupilot/pull/12642

mohanbabi, 2021-03-01:
Hi Sir, We can achieve this resume of the mission.
1.DISARM_DELAY=0
2. RCX_OPTION(better to use 2 position switch)=31

1.Plan the mission in the drone. interrupt the mission by changing flight mode. then land the drone, after land, switch on **Motor Emergency Stop**(switch High position) and load some require(i.e payload).
2.Release the **Motor Emergency Stop**(Switch low position), take off, and change the mode to auto. 
I have tried this procedure for mission resuming for agriculture sprayer drones.

Tip: - Make landing place as flate. (This will use for avoiding auto disarming drones.) if the drone is disarmed mission is reset to the initial.
If anything is wrong excuse me.

Thanks and regards
Mohan

IamPete1, 2021-08-15:
Sounds like this is fixed, https://github.com/ArduPilot/ardupilot/pull/12642#issuecomment-879687938

closed 2021-08-15 by IamPete1