ArduPilot/ardupilot #12661: Copter/Rover: revisit object database object sizes

rmackay9, 2019-10-26:
During testing of BendyRuler on Copter-4.0, the latest version of MP displayed objects as shown below with many circles of widely varying sizes that give the user very little indication as to where objects are.

I think we should probably simplify the calculations used in the OADB to a value that is simply relative to the beam width x distance.  Open to other suggestions as well though of course.  @magicrub, @meee1.

[image]

WickedShell, 2019-10-26:
Vector2I for the location, and we can add an altitude back later if needed.

Also this will save 4 bytes (32 -> 28 bytes per DB item)

```
diff --git a/libraries/AC_Avoidance/AP_OADatabase.h b/libraries/AC_Avoidance/AP_OADatabase.h
index 1d2f37dac8..971e3ac89d 100644
--- a/libraries/AC_Avoidance/AP_OADatabase.h
+++ b/libraries/AC_Avoidance/AP_OADatabase.h
@@ -21,7 +21,7 @@ public:
         return _singleton;
     }
 
-    enum OA_DbItemImportance {
+    enum OA_DbItemImportance : uint8_t {
         Low, Normal, High
     };
 
```

rmackay9, 2019-10-26:
@WickedShell, thanks for the advice.  Sounds reasonable.  This is not really what this issue is related to though.  It's not the size in memory but rather the size recorded for each object in the database.  I.e. each object has a location and a size.  The way we decide on the sizes can be improved I think.

WickedShell, 2019-10-26:
@rmackay9 I'm fuzzy on it now, but don't we start increasing the size because we ran out of space for them? IE I thought smaller items meant a larger DB which leads to less size increasing.

rmackay9, 2019-10-26:
@WickedShell, I'm not sure how we could run out of space when [we're using a float](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADatabase.h#L31).  Maybe I should stop saying "size" and start saying "radius".

WickedShell, 2019-10-26:
Oh I meant space as in the radiuses on objects get increased to include other things to avoid because we need the object to represent both of them. So my comment was less that the float was a problem, but we can't support enough DB items to allow the radiuses to stay small.

magicrub, 2019-10-27:
Right, let's use radius instead of size. Michael had a point though. The radius is a function of db fullness. Bigger db capacity means more smaller circles. So, by reducing the db item memory footprint then the default capacity can be larger, this giving smaller circles.

Also, current max radius is 10m. That's probably too big. 

I propose:
- reduce max radius to 5m.
- remove alt from database per MdB's suggestion.
- increase default db size from 100 to 200.

magicrub, 2019-10-27:
Also, circles should not be overlapping. This is a downside of the current gcs output implemention that just send them in index order instead of being event driven.

closed 2020-01-01 by rmackay9