ArduPilot/ardupilot #12663: Rover: blank pre-arm message

rmackay9, 2019-10-26:
A newly setup vehicle may display a blank prearm message, "PreArm:".  I don't know what the cause is but we should resolve this.

This is probably the same issue as I've found for TradHeli: https://github.com/ArduPilot/ardupilot/issues/11800

peterbarker, 2019-10-26:
On Fri, 25 Oct 2019, Randy Mackay wrote:

> A newly setup vehicle may display a blank prearm message, "PreArm:". I don't know what the cause is but we should resolve this.
> 
> This is probably the same issue as I've found for TradHeli: #11800

Rover may have been branched before I patched this with 
3619856fc9bac3744d83c7b9a0127e0f1f56f667 and 
1c187d8feb78a1c71311baa8458f52b6cccd6dba ?

rmackay9, 2019-10-28:
We've also got this PR that fixes a blank message: https://github.com/ArduPilot/ardupilot/pull/12651

rmackay9, 2019-11-01:
This is fixed in master so closing

closed 2019-11-01 by rmackay9