ArduPilot/ardupilot #12682: add a  pure GPS-RTH mode for quadcopter, do not need compass

sin5678, 2019-10-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

### Feature request

**Is your feature request related to a problem? Please describe.*
make the quadcopter fly like fixwing , only need GPS to do RTH ,better have basic GPS hold

**Describe the solution you'd like**
when do RTH , quad tilt to an pre-define angle ,and fly forward ,then we known the heading info from the GPS , then do a small yaw control , till we heading home point , just like fix-wings. better disable roll control at the same time , we do not need roll control  now and  too much roll may cause one motor overload in big wind.  no compass ,no more compass calibration to do ,for experienced pilot, it's totally ok and feel easy to build a quad (no more compass interference),

**Describe alternatives you've considered**

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

rmackay9, 2021-06-01:
It is possible that the GSF feature from 4.1 will resolve this issue because it allows discovering the yaw by comparing the GPS and IMU values.

IamPete1, 2021-09-12:
This is possible in 4.1 using GSF, recommended that you have responsibly good GPS.

closed 2021-09-12 by IamPete1